platformio/platformio-core #4310: Quoting rules for `build_flags` (and such) are subtle, need better documentation

egnor, 2022-06-08:
What kind of issue is this?

- [✔️] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 22.04 LTS (Linux 5.15.0)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem

When messing with `build_flags` and other flags added to build programs, one might need to quote characters like spaces or quotes. For example, to define a macro that expands to a C quoted string which includes quotation marks, one would do something like this in `platformio.ini`:

```
build_flags =  '-DMYMACRO="My \\"Quoted\\" String"'
```

(On the command line shown by `pio run -v`, this turns into `-DMYMACRO=\"My \\\"Quoted\\\" String\""`.) This is not at all obvious, I had to spend a while experimenting to figure out the rules, and I'm still quite confused about the rules, which seem to depend on which exact flags are being used. The flow seems to be something like this:

1. A generic `.ini` file parser runs, with its own quoting rules to prevent values from being split up
2. Some logic runs that attempts to parse flags, interpreting quotes the same-ish way a shell would?
3. Flags are sorted and rearranged and put back on a command line, then requoted for the shell's benefit
4. Then the shell parses the flags, hopefully the expected way
5. Then the compiler interprets the arguments, hopefully the expected way

I am definitely not sure how the 2nd and 3rd steps work. For example, if I specify this:

```
build_flags = '-DTEST1 "-DTEST2=Test2 String"'
```

the compiler command ends up like this:

```
arm-none-eabi-g++ ... "-DTEST1 "-DTEST2=Test2 String\""
```

... which causes the shell to (correctly) say `Syntax error: Unterminated quoted string`. That feels like a bug somewhere? Or, perhaps, a very subtle syntax, which definitely ought to be explained better. I'm happy to make a PR against the documentation, but I need to understand what the rules actually are, first!

roipoussiere, 2022-06-10:
I totally agree, it's confusing in particular when working with dynamic build flags.

It took me a couple of hours to figure out that the correct way to include the commit hash in a build argument without calling a custom script was:

```
build_flags =
  !echo '-D COMMIT=\\"'$(git rev-parse --short HEAD)'\\"'
```

closed 2022-06-17 by ivankravets

ivankravets, 2022-06-17:
Thanks for the issue! See updated docs https://docs.platformio.org/en/latest/projectconf/section_env_build.html#stringification

bergerb-com, 2024-10-08:
For Windows:

git-version.cmd:
```
@echo off

FOR /F "delims=" %%i IN ('git rev-parse --short HEAD') DO set hash=%%i
echo '-D COMMIT_HASH="%hash%"'
```

```
build_flags = !git-version.cmd
```

commit d4784c05 (2022-06-17): Documented Stringification – converting a macro argument into a string constant  // Resolve #4310