platformio/platformio-core #4312: Raspberry pi pico multicore

SebMoy, 2022-06-08:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 - Visual studio code 1.67.2

**PlatformIO Version** (`platformio --version`): PlatformIO 2.4.3

### Description of problem
I would like programming the raspberry pico in multicore.

I program in PlatformIO is not execute.
I program in Arduino is run.

#### Steps to Reproduce
The code https://github.com/earlephilhower/arduino-pico/blob/master/libraries/rp2040/examples/Multicore/Multicore.ino 

### Actual Results
I programming in Arduino.....

Thank you.

maxgerhardt, 2022-06-09:
Working when using my adapted platform per https://arduino-pico.readthedocs.io/en/latest/platformio.html. When creating a regular Arduino + Pico project in PlatformIO uses the mbed core, which works totally different and doesn't have that multicore support. 

This is already tracked in https://github.com/platformio/platform-raspberrypi/issues/13

ivankravets, 2022-06-09:
Duplicate of https://github.com/platformio/platform-raspberrypi/issues/13

closed 2022-06-09 by ivankravets

SebMoy, 2022-06-09:
Thank you for answer.

I did not see the issues https://github.com/platformio/platform-raspberrypi/issues/13