platformio/platformio-core #4313: Fallback for upload/monitor ports

guestisp, 2022-06-10:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

Would be useful to add a sort of fallback in `upload_port` and `monitor_port`, something like:

```
upload_port = /dev/ttyUSB0 | /dev/ttyUSB1
monitor_port = /dev/ttyUSB0 | /dev/ttyUSB1
```

should the first one fail , the second one is used (or the third, or the fourth and so on).

Asking this because we develop on multiple workstations and not all ports are the same on every workstation. Curretly we have to change the port manually (or pointing to a custom platformio.ini). A fallback would fix this on it's own.

ivankravets, 2022-06-10:
Have you tried patterns?

```
upload_port = /dev/ttyUSB[01] 
```

guestisp, 2022-06-10:
yes, if the first one exists but doesn't work (ie because is not an arduino) uploads fails and Pio doesn't try the next one. 

my suggestion is about to try the next one if firsr fails, patterns is useful to make a list of available serials but only the First one is used

ivankravets, 2022-06-10:
> yes, if the first one exists but doesn't work

Is it possible? If you have the correct VID/PID, PlatformIO will automatically detect port. What is your board? If this is Chinese noname board, please configure https://docs.platformio.org/en/latest/scripting/examples/override_board_configuration.html

Does it work now?

guestisp, 2022-06-10:
i have 2 or 3 different esp32, Arduino Uno (elegoo), esp8266 and so on.

in all cases, as ttyUSB0 is connected to a different device, the proper Port is ttyUSB1, but pio tries to write on usb0, if fails then exit

ivankravets, 2022-06-10:
Good. Let's see if someone else need this feature voting using 👍  in the original comment.

I don't think this feature is useful, only adds complexity and break compatibility because upload_port always was singular.

If there are no enough votes, we will close it. Please use project environment and configure board PID/VID. `pio run` and `pio device monitor` support `-e`.

tobozo, 2022-06-27:
Hello :wave: 

I can confirm the serial port finder has been significantly improved in `6.0.3rc2`, here's what I had before upgrading:

- Platformio 6.0.2
- No `upload_port` or `monitor_port` directive in platformio.ini
- Linux desktop with Conbee dongle on `/dev/ttyACM0` and M5Stack on `/dev/ttyUSB0`

Test: `pio run --target=upload --upload-port=/dev/ttyUSB0 --target=monitor`

Expected result: after flashing the M5Stack and before entering monitor mode, platformio should ignore the Conbee port and wait for `--upload-port` to become available as the monitor port

Actual result: platformio doesn't see `/dev/ttyUSB` while it's being reset, and wrongly enables monitor mode on the Conbee port instead of the M5Stack port

`pio upgrade --dev` fixed that, thanks for the hard work!

ivankravets, 2022-06-27:
Thanks for confirming that it works as expected now. Please post your comments if you have any issues, we will fix them ASAP.

The PIO Core 6.0.3 is going to be released this week.

closed 2022-06-27 by ivankravets