platformio/platformio-core #4315: Trying to add new library

piapiekarek, 2022-06-12:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration
Newly installed 

**Operating system**:
Windows 11 (same with Windows 10)

**PlatformIO Version** (`platformio --version`):
6.0.2

Python version: 3.9.10

### Description of problem
I want to add a library in a new project (VSCode and PlatformIO are newly installed)
PS C:\Users\piapi\Documents\PlatformIO\Projects\HID-FootSwitch> pio pkg install --library "arpruss/USBComposite for STM32F1@^1.06"
Resolving bluepill_f103c8 environment packages...
Error: Traceback (most recent call last):
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 65, in package_install_cmd
    install_project_dependencies(options)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 106, in install_project_dependencies
    already_up_to_date = not install_project_env_dependencies(env, options)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 126, in install_project_env_dependencies
    _install_project_env_custom_libraries(project_env, options)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 300, in _install_project_env_custom_libraries
    spec = PackageSpec(library)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 143, in __init__
    self._parse(self.raw)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 250, in _parse
    raw = parser(raw)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 275, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 190, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\piapi\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block '^1.06'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

#### Steps to Reproduce

install everything anew and try adding a lib?

### Actual Results
Above error description

### Expected Results
A library added to the project? :D

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Additional info
I have used PlatformIO before on Linux without problems (but I don't have Linux installed anymore).
I tried using docker to avoid the issue but I get the same results...

ivankravets, 2022-06-13:
Please forward the issue to https://github.com/arpruss/USBComposite_stm32f1/issues. 
The `1.06` is not a valid [semantic version](https://semver.org/). It should be 1.6.

The `pio pkg install --library "arpruss/USBComposite for STM32F1@^1.6"` works for me.

ivankravets, 2022-06-13:
No need to be sorry. You are right, there is a bug but we can't fix it on our side. Please ask the library maintainers to release a new version under `1.7` tag.