platformio/platformio-core #4317: Intellisense broken on Unix-like OSes and new C/C++ extension version

maxgerhardt, 2022-06-12:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.0.4 LTS

**PlatformIO Version** (`platformio --version`): 6.0.3a1

### Description of problem

Intellisense broken in VSCode for builde scripts that use `-iprefix <path> @<include file>`-style includes.

#### Steps to Reproduce

1. Use above OS (or any Linux OS or Mac) with provided `platformio.ini`
2. Open project in VSCode
3. Build project (successfull)
4. Observer Intellisense 

### Actual Results

Red error lines for intellisense.

### Expected Results

Intellisense finds all include paths.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:pico]
platform = https://github.com/platformio/platform-raspberrypi.git
board = pico
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() { }
void loop() {
}

```

### Additional info

Originally thought to be bug in the builder script that exposes the include paths (in `env["CPPPATH"]`), but then discovered to be dependent on the Microsoft C/C++ extension version. See https://github.com/platformio/builder-framework-arduino-core-mbed/issues/4. Tracked in the core because of high-impact.

maxgerhardt, 2022-06-12:
Tracked on the VSCode side in https://github.com/microsoft/vscode-cpptools/issues/9434.

valeros, 2022-06-15:
Hi @maxgerhardt , thanks for reporting. Should be fixed in the latest VSCode C/C++ extension [v1.10.6](https://github.com/microsoft/vscode-cpptools/releases/tag/v1.10.6).

closed 2022-06-15 by valeros