platformio/platformio-core #4318: How does platform upload Download bin to the address of 0x8008000?

solawc, 2022-06-13:
Hello, I have built the platform IO project of STM32 from scratch. The program runs well, and there is no problem in compiling and downloading. Then I need to write a bootloader, but I find that if I press F5 to execute debugging, the code will be downloaded to me The address of 0x8008000 set on the LD file is correct. But I directly download the program by pressing the upload button. It will download the program from the address of 0x8000000 until the download is completed. My bootloader is covered in the middle. I've been trying to find the relevant offset settings in the document, but I can't find them. Can you tell me what I can do?

maxgerhardt, 2022-06-13:
https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L312-L315

```ini
board_upload.offset_address = 0x8008000
```

solawc, 2022-06-13:
> 
[image]

Thank you very much for your answer. This is the information obtained after modification. I can see from the information that it still starts from 0x8000000 instead of 0x08008000. Did I miss any information? Or did I get it wrong?

maxgerhardt, 2022-06-13:
This is also supposed to work with JLink: 

https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L173-L188

In the `.pio\build\<env>` folder, there should be a `upload.jlink`. What's its content?

maxgerhardt, 2022-06-13:
I think you're interpreteing the output incorrectly, it just says that "Bank 0" of flash starts at 0x08000000, which is absolutely correct. It should still be using the correct address, and if it's smart, it should not erase the previous bytes (if it's sector aligned).

solawc, 2022-06-13:
Oh, yes, I really misunderstood it. He has been able to work normally. Thank you.

solawc, 2022-06-13:
> I think you're interpreteing the output incorrectly, it just says that "Bank 0" of flash starts at 0x08000000, which is absolutely correct. It should still be using the correct address, and if it's smart, it should not erase the previous bytes (if it's sector aligned).

Oh, yes, I really misunderstood it. He has been able to work normally. Thank you.