platformio/platformio-core #4319: monitor_speed directive ignored on Core 6.0.3a3

thomastech, 2022-06-13:
After today's update to core 6.0.3a3 (was 6.0.3a2) the monitor_speed entry in platformio.ini is ignored. Instead baud defaults to 9600.

Baud can still be changed using the terminal command
`pio device monitor --baud 115200`

More Info:
I reverted to 6.0.2 and monitor_speed was working again. Next I tried to re-update to the latest 6.0.3 release but the terminal command line unexpectedly reports that 6.0.2 is the latest version. I expected it to update me to the very latest 6.0.3 branch with the hope this issue had been resolved. 

Configuration:
Win10 Pro, Platformio home 3.4.1, core 6.0.3a3

#### Steps to Reproduce
1. Update to core 6.0.3a3
2. Set _monitor_speed = 115200_ in platformio.ino
3. Launch monitor. 

### Actual Results
monitor baud 9600.

### Expected Results
monitor baud 115200

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
; Revised Jun-13-2022, TEB.
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = ESP32
src_dir = ./
data_dir = ./data
build_cache_dir = ~/.buildcache
extra_configs =
	platformio_user.ini

[env:ESP32]
platform = espressif32 @ ^4.4.0
framework = arduino
board = ttgo-t1 ; Use this board profile because TTGO-T8 not defined yet.
board_build.filesystem = littlefs
board_build.flash_mode = dout
;check_tool = clangtidy
check_tool = cppcheck ; ALternate Inspector, use CLI: pio check --skip-packages
check_flags = --enable=all
check_severity = medium, high
check_skip_packages = yes
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
upload_speed = 921600
board_build.f_cpu = 240000000L
board_build.partitions = min_spiffs.csv
board_upload.flash_size = 4MB
board_upload.maximum_ram_size = 327680

build_type = release
;build_type = debug
build_flags =
	-DCORE_DEBUG_LEVEL=0 ; Release=0, Set to 5 for debugging.

lib_deps =
	PubSubClient @ ^2.8
    LittleFS @ ^2.0.0
    ;me-no-dev/ESP Async WebServer @ ^1.2.3 ; Old Lib, not compatible with latest core.
    ;https://github.com/me-no-dev/ESPAsyncWebServer.git ; Updated lib, working.
    https://github.com/esphome/ESPAsyncWebServer @ ^2.1.0 ; Updated lib, seems to have recent patches.
	me-no-dev/AsyncTCP @ ^1.1.1
	thijse/ArduinoLog @ ^1.1.1
	lbernstone/Tone32 @ ^1.0.0
    bblanchon/ArduinoJson @ ^6.19.4

extra_scripts =
	.scripts/LittleFSBuilder.py
```

**The content of `platformio_user.ini`:**
```ini
; This add-on Platformio ini file is for the project builder's environmental directives.
[env:ESP32]
upload_port = COM8
monitor_port = COM8
monitor_rts = 0
monitor_dtr = 0
```

bakeddean, 2022-06-13:
Same issue. How did you revert to 6.0.2?

thomastech, 2022-06-14:
> Same issue. How did you revert to 6.0.2?

I fumbled around and used a couple different commands in the IDE's Terminal window.
I could be wrong but I think the correct command is:
`pip install --upgrade pip`

_pio -- version_ reports I'm now on release 6.0.2. Which is the latest. But now I often see this message

```
**************************************************************************************************
Obsolete PIO Core v6.0.2 is used (previous was 6.0.3a2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
**************************************************************************************************

```
Despite following the instruction to fix the Obsolete PIO Core warning, the message continues to show up. A visual nuisance, but does not affect builds. I suspect it will go away when PlatformIO IDE updates to the next official release. Fingers crossed.

- Thomas

bakeddean, 2022-06-14:
Thanks. I was looking at this [https://github.com/platformio/platformio-core/blob/develop/platformio/commands/upgrade.py](platformio/commands/upgrade.py), and ended up downloading the 6.0.2 zip and doing:
```
sudo pip install --upgrade ~/Documents/platformio-core-6.0.2.zip
```
Pretty much the same as you but I didn't do an uninstall. I got the warning also when doing a build.
I went into my .platformio folder ```/Users/dean/.platformio``` (on a Mac), and opened ```appstate.json``` and changed ```"last_version": "6.0.3a2"``` to ```"last_version": "6.0.2"``` and the warning disappeared.

thomastech, 2022-06-14:
> I went into my .platformio folder /Users/dean/.platformio (on a Mac), and opened appstate.json and changed "last_version": "6.0.3a2" to "last_version": "6.0.2" and the warning disappeared.

Thanks for that tip. The nuisance warning is now gone.

- Thomas

bakeddean, 2022-06-14:
👍

ivankravets, 2022-06-14:
Sorry, this is a regression bug. Could you re-test with `pio upgrade --dev`? 

We made huge refactoring to the device monitor. It now handles disconnects and provides better stability.

flyback88, 2022-06-14:
This upgrade to 6.0.3a3 works for the speed, thanks.
But it does not work for the filters "direct" and the flags "--raw" both needed to handle ANSI color codes. Look likes those options are not taken into account by Monitor.

DA.

ivankravets, 2022-06-14:
Works for me, [try this project](https://github.com/platformio/platformio-core/files/8901207/111.zip)

[image]

commit bd21ff0d (2022-06-14): Fixed an issue when monitor_speed was ignored in configuration file // Resolve #4319