platformio/platformio-core #4320: add timestamp to logfiles

HunterTom94, 2022-06-14:
What kind of issue is this?

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Description of problem
I know by monitor_filters= time, log2file, you can show timestamp in the terminal and log serial output to a file. However, is it possible to also include the timestamp in the logged file?

**The content of `platformio.ini`:**
```ini
monitor_filters= time, log2file
```

ivankravets, 2022-06-15:
Thanks for the issue. Could you re-test the latest development branch? 

```
pio upgrade --dev
```

Does it work now?

closed 2022-06-15 by ivankravets

HunterTom94, 2022-06-15:
Hi,

The timestamps are now showing up but I got something else mixed in my
message.

The content in the serial monitor is fine. However, in the log file, I got
something below (the last few lines are fine and are the expected output).

Also, do you know how to revert back to the non-dev branch? If I type in "pio
upgrade" it will say "You're up-to-date! PlatformIO 6.0.3a4 is currently
the newest version available."

Thank you!

[37m12:29:08.947 > Rea [37mch [37med [37m [37mT [37mar [37mg [37me [37mt
[37m P [37mo [37ms [37mi [37mt [37mio [37mn [37m: [37m 0 [37m.1 [37m5
[37m
[37m12:29:09.179 > C [37murrent [37mStatus [37m: idl [37me, Po [37mtent
[37miomete [37mr Pos [37mitio [37mn: 18 [37m6.00, [37m0.15 [37m, Tar
[37mget: [37m 185, [37m 0.1 [37m5, Bri [37mghtnes [37ms: 108 [37m, Curr
[37ment Fo [37mrce: 0 [37m.00 N
[37m
[37m12:29:09.443 > C [37murrent [37m Statu [37ms: idl [37me, Po [37mtenti
[37mometer [37m Posi [37mtion: [37m 186. [37m00, 0 [37m.15, [37mTarg
[37met: [37m185, [37m0.15 [37m, Br [37might [37mness: [37m106, [37m Curr
[37ment [37mForce: [37m 0.0 [37m0 N

[37m12:29:09.708 > C [37murrent St [37matus: [37midle, [37mPoten [37mtiome
[37mter P [37mosit [37mion: [37m186. [37m00, [37m0.15, [37m Tar [37mget:
[37m 185, [37m 0.1 [37m5, B [37mright [37mness [37m: 109, [37m Curre [37mnt
Fo [37mrce: [37m 0.00 [37m N

[37m12:29:09.971 > C [37murrent Status: idle, Potentiometer Position: 186.00,
0.15, Target: 185, 0.15, Brightness: 109, Current Force: 0.00 N

12:29:11.140 > Current Status: idle, Potentiometer Position: 186.00, 0.15,
Target: 185, 0.15, Brightness: 106, Current Force: 0.00 N

12:29:11.140 > Current Status: idle, Potentiometer Position: 186.00, 0.15,
Target: 185, 0.15, Brightness: 105, Current Force: 0.00 N

ivankravets, 2022-06-15:
Could you try to add default filter? See docs https://docs.platformio.org/projectconf/section_env_monitor.html

What is your hardware? It could be a bootloader output.

HunterTom94, 2022-06-15:
Interestingly, adding the "default" filter caused that " [37m" to also show
up in the serial monitor. If I remove the "default" filter, it only shows
up in the log file.

Also I don't understand what the "default" filter means by "Remove typical
terminal control codes from input".

Also my device is Arduino Nano Every

ivankravets, 2022-06-15:
It looks like a garbage from boot-loader. You can check it with `hexlify` filter. But please remove all other filters. 

Reverting back to 6.0.2. See installation guide (docs) for PIO Core.

commit 6f998512 (2022-06-15): Fixed an issue when the monitor filters were not applied in their order // Resolve #4320