platformio/platformio-core #4326: error "complete: nosort: invalid option name" after installing bash completion

RobertoD91, 2022-06-19:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
MacOS 12.4

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.0.2

### Description of problem
After `pio system completion install bash` when i open a new terminal i get the error `-bash: complete: nosort: invalid option name` and autocomplete for pio don't work

#### Steps to Reproduce

1. run `pio system completion install bash`
2. open a new terminal

### Actual Results
```
$ pio system completion install bash
PlatformIO CLI completion has been installed for BASH shell to /Users/Roberto/.bash_completion 
Please restart a current shell session.
```

### Expected Results
...

### Additional info
Bash version: `3.2.57(1)-release`

RobertoD91, 2022-06-19:
with bash version `5.1.16(1)-release` installed from homebrew it work

ivankravets, 2022-06-20:
Thanks, fixed! Please re-test with `pio upgrade --dev`.

See https://click.palletsprojects.com/en/8.1.x/shell-completion/#shell-completion

commit 092326cb (2022-06-20): Warn about incompatible Bash version for the Shell Completion // Resolve #4326
closed 2022-06-20 by ivankravets