platformio/platformio-core #4327: Invalid simple block 'v0.1.0'

RodrigoDornelles, 2022-06-20:
What kind of issue is this?

- [X] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [X] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

```Darwin mac.local 19.6.0 Darwin Kernel Version 19.6.0: Mon Apr 18 21:50:40 PDT 2022; root:xnu-6153.141.62~1/RELEASE_X86_64 x86_64```

**PlatformIO Version** (`platformio --version`):

```
PlatformIO Core, version 6.0.2
```

### Description of problem

Error when require deps

#### Steps to Reproduce

1. put deps on plataformio.ini
```
lib_deps =
    https://github.com/rodrigodornelles/3bc-lang
```

2. click on build

### Actual Results

```
Removing unused dependencies...
Error: Traceback (most recent call last):
  File "/Users/rodrigo/.platformio/penv/lib/python3.9/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/rodrigo/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/rodrigo/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/rodrigo/.platformio/penv/lib/python3.9/site-packages/platformio/cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "/Users/rodrigo/.platformio/penv/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
[... 141 lines cut ...]
[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps =
    https://github.com/rodrigodornelles/3bc-lang
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <3bc.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Additional info

I'm the maintainer of the library I'm trying to include.

ivankravets, 2022-06-20:
> ValueError: Invalid simple block 'v0.1.0'

Please remove broken dependencies that do not conform to https://semver.org/

RodrigoDornelles, 2022-06-20:
> Could you try to remove `.pio` folder from the project and build again?

It worked