platformio/platformio-core #4330: [ERROR]: Unknown board ID 'lolin_s2_mini'

hasenradball, 2022-06-25:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ x] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = arduino

monitor_speed = 115200
upload_protocol = esptool
upload_speed = 921600
upload_port = COM6
```

**Operating system**:

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.0.2

### Description of problem
When starting to build it immediately aborted with error:

```
> Executing task: C:\Users\Frank\.platformio\penv\Scripts\platformio.exe run <

Processing lolin_s2_mini (platform: espressif32; board: lolin_s2_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------

Error: Unknown board ID 'lolin_s2_mini'
Der Terminalprozess "C:\Users\Frank\.platformio\penv\Scripts\platformio.exe 'run'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

```

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = arduino

monitor_speed = 115200
upload_protocol = esptool
upload_speed = 921600
upload_port = COM6

```

**Operating system**:

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.0.2

### Description of problem
When starting to build it immediately aborted with error:

```
```

**Source file to reproduce issue:**
```
#include <Arduino.h>
#include <WiFi.h>

// ============================================
// ### --- START: Klassen Instanzen --- ###
// ============================================

// ============================================
// ### --- ENDE: Klassen --- ###
// ============================================

void setup() {
  WiFi.mode(WIFI_OFF);
  // put your setup code here, to run once:
  Serial.begin(115200);
  while(!Serial) yield();
  Serial.println();
}

void loop() {
}
```

### Additional info

ivankravets, 2022-06-26:
See supported boards https://registry.platformio.org/platforms/platformio/espressif32/boards

closed 2022-06-26 by ivankravets

hasenradball, 2022-06-26:
Thanks for your fast reply.
I am actually not so expirieced with the Esp32 and its platforms

what is the difference beteeen:

platform = espressif32

or

platform = platformio/espressif32@….

Are this different repositories?

ivankravets, 2022-06-27:
Please remove `C:\Users\Frank\.platformio\platforms` folder and try building the project again. 

The `platform = espressif32` matched to any dev-platform that has a name=espressif32. If you didn't install any ESP32 dev-platorms before, it will match the official https://registry.platformio.org/platforms/platformio/espressif32

The `platform = platformio/espressif32` matches strictly the official dev-platform maintained by PlatformIO Labs company exclusively for Espressif company.

ivankravets, 2022-07-07:
Yes, the same for any dev-platform. This is the reason why we recommend to install official dev-platforms.