platformio/platformio-core #4331: Clean All fails on files that are read only

alex-code, 2022-06-26:
- [x] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11

**PlatformIO Version** (`platformio --version`): 6.0.2

### Description of problem
**Clean All** fails on files that are "**Read Only**".
Including a git repository in **lib_deps** some files in the `.git` directory are marked read only which blocks cleaning.

#### Steps to Reproduce

1. Have a project include https://github.com/PaulStoffregen/Encoder.git in **lib_deps**
2. Build or let the libs be downloaded
3. Run **Clean All**

### Actual Results
```
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\myuser\\Documents\\GitHub\\My-Project\\.pio\\libdeps\\firebeetle32\\Encoder\\.git\\objects\\pack\\pack-87f987569879568m596f8m96f8fjddl0u.idx':
```

### Expected Results
The files to be removed.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:firebeetle32]
monitor_speed = 115200
platform = espressif32
board = firebeetle32
framework = arduino
monitor_filters = esp32_exception_decoder
extra_scripts = extra_script.py
board_build.partitions = no_ota.csv
lib_ldf_mode = chain+
lib_compat_mode = strict
lib_deps = 
	https://github.com/PaulStoffregen/Encoder.git
```

### Additional info

Adding this line `os.chmod(dst, stat.S_IWRITE)` to the `piotarget.py` file just before the `remove` allows the clean to run successfully but I'm not sure if this is the best way to fix it.

ivankravets, 2022-06-27:
Thanks for the report. Please re-test with `pio upgrade --dev`. Does it clean without errors now?

closed 2022-06-27 by ivankravets

alex-code, 2022-06-27:
All good, clean completes successfully.
Thanks for updating 👍

commit 99b52048 (2022-06-27): Fixed a "PermissionError" on Windows when running "clean" or "cleanall" targets // Resolve #4331