platformio/platformio-core #4334: extra_scripts pre:  import( "env", "projenv" ) not working/defined

tcpiinterface, 2022-06-27:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem

extra_scripts pre:  import( "env", "projenv" ) not working/defined variables

#### Steps to Reproduce

Create new project with provided files, try to build

### Actual Results

Erro messages

*** Import of non-existent variable ''projenv''
File "script_pre.py", line 1, in <module>

### Expected Results

Have projenv variables available to script, before build  event

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = Default

[env:Default]
platform = espressif32@^4.4.0
framework = arduino
board = esp32dev
extra_scripts =
  pre:script_pre.py
  post:script_post.py
```

**Source file to reproduce issue: script_pre.py** Not Works!
```
Import("env", "projenv")
proj_dir = env.get('PROJECT_DIR')
if env.GetProjectOption("build_type") == "debug":
    print( "debug" )
```
**Source file to reproduce issue: script_post.py** Works!
```
Import("env", "projenv")
proj_dir = env.get('PROJECT_DIR')
if env.GetProjectOption("build_type") == "debug":
    print( "debug" )
```

### Additional info

No matther if scripts have more code, always raise build error.

tcpiinterface, 2022-06-27:
Upgraded.. but still same as before.

Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 6.0.3rc2

ivankravets, 2022-06-27:
`projenv` is not available in the `PRE` type script.

Tarek-Ragab-Abdelal, 2023-04-30:
@tcpiinterface , Can you please tell me if you figured out how to resolve this problem?