platformio/platformio-core #4338: Library dependencies installing for all platforms

honzakucera, 2022-06-28:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Linux/Ubuntu 22.04

**PlatformIO Version** (`platformio --version`):
Tested in 6.0.2 and 6.1.0b1

### Description of problem
For some reason, PlatformIO ignores dependency platforms when installing dependencies from library.json from an actual dependency defined in platformio.ini.

My platformio.ini
```ini
[platformio]
default_envs = esp32dev

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	ottowinter/ESPAsyncWebServer-esphome @ ^2.1.0
board_build.f_cpu = 80000000L
monitor_speed = 115200
```

Library.json of one of the affected libraries
```json
{
  "name":"ESPAsyncWebServer-esphome",
  "description":"Asynchronous HTTP and WebSocket Server Library for ESP8266 and ESP32",
  "keywords":"http,async,websocket,webserver",
  "authors":
  {
    "name": "ESPHome Team",
    "maintainer": true
  },
  "repository":
  {
    "type": "git",
    "url": "https://github.com/esphome/ESPAsyncWebServer.git"
  },
  "version": "2.1.0",
  "license": "LGPL-3.0",
  "frameworks": "arduino",
  "platforms": ["espressif8266", "espressif32"],
  "dependencies": [
    {
      "owner": "ottowinter",
      "name": "ESPAsyncTCP-esphome",
      "platforms": "espressif8266"
    },
    {
      "owner": "esphome",
      "name": "AsyncTCP-esphome",
      "platforms": "espressif32"
    },
    {
      "name": "Hash",
      "platforms": "espressif8266"
    },
    {
      "name": "ESP8266WiFi",
      "platforms": "espressif8266"
    }
  ]
}
```

According to PlatformIO docs, PlatformIO should only install dependencies for the given platform. Instead, it appears that it installs all of them regardless on the platform.

Output of a fresh build using these:
```
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Library Manager: Installing ottowinter/ESPAsyncWebServer-esphome @ ^2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing Hash
Library Manager: Installing ESP8266WiFi
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 80MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|   |-- ESP8266WiFi
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- AsyncTCP-esphome @ 1.2.2
|-- ESP8266WiFi
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- WiFi @ 2.0.0
```

#### Steps to Reproduce

1. Create a new project for the espressif32 platform
2. Copy platformio.ini from the example given above

### Actual Results
All library dependencies get installed regardless on the platform.

### Expected Results
Only the dependencies for the platform used should be installed

closed 2022-07-04 by ivankravets

alex-code, 2022-07-04:
Looks good to me 👍

honzakucera, 2022-07-05:
Works perfectly. Thank you for the quick fix :+1: