platformio/platformio-core #4339: Doctest is fairly verbose when a test fails

jcw, 2022-06-29:
What kind of issue is this?

- [x] **Feature Request**.

------------------------------------------------------------------

### Configuration

MacOS, PIO 6.0.3rc2.

### Description of problem

The `doctest` framework can generate a lot of output when a single test fails.
This is because all successful tests are _also_ reported, due to this setting in `main()` (as required by PIO):

    context.setOption("success", true);     // Report successful tests

My hunch is that PIO parses the output to generate its test-cases/succeeded report at the end.
If this is so, then perhaps it could be solved by instead parsing doctest's own report at the end, e.g.
```
[doctest] test cases:  5 |  5 passed | 0 failed | 0 skipped
[doctest] assertions: 95 | 95 passed | 0 failed |
[doctest] Status: SUCCESS!
```

### Additional info

For non-trivial test suites, the error message can end up buried anywhere inside that non-colorised final PIO output, making it hard to figure out what failed. Here's an example:
```
test/basic.cpp:55:
TEST CASE:  ByteVec
  insert-relpos

test/basic.cpp:88: SUCCESS: CHECK( v.size() == 10 ) is correct!
  values: CHECK( 10 == 10 )

test/basic.cpp:89: SUCCESS: CHECK( v.begin() != nullptr ) is correct!
  values: CHECK( 0x600002ce0060 != nullptr )

test/basic.cpp:90: SUCCESS: CHECK( v.capacity() >= 10 ) is correct!
  values: CHECK( 10 >= 10 )

test/basic.cpp:92: SUCCESS: CHECK( v.relpos(3) == 3 ) is correct!
  values: CHECK( 3 == 3 )

test/basic.cpp:93: ERROR: CHECK( v.relpos(-3) == 6 ) is NOT correct!
  values: CHECK( 7 == 6 )

test/basic.cpp:97: SUCCESS: CHECK( v[3] == 11 ) is correct!
  values: CHECK( 11 == 11 )

test/basic.cpp:98: SUCCESS: CHECK( v[-3] == 22 ) is correct!
  values: CHECK( 22 == 22 )

=============== 16 test cases: 1 failed, 15 succeeded in 00:00:01.282 ===============
```

(my actual test source code is here: https://git.jeelabs.org/doodle/tree/dsman/test/basic.cpp)

For comparison, this is a test failure in another non-PIO doctest project, with doctest's default main:

[image]

PS. This issue only applies to native builds (since doctest doesn't currently support embedded builds).
It's also not _that_ bad, since only the checks in the _failing_ `TEST_CASE` are shown, not all test cases.
IOW, it's not _terribly_ important to act on this, I just wanted to report my surprise & slight inconvenience.

ivankravets, 2022-07-01:
So, and what could we do here? We keep a unified API per all supported testing frameworks. It allows developers to export PlatformIO testing results into JSON/XML no matter which framework they used. This is the reason why we CAST different testing frameworks to PlatformIO's unified results API interface.

I agree, we could have a case where the developer does not like our unified behavior. In this case, we document how to write your own test runner and output in format which is comfortable to the developer. See docs https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/custom/index.html

Could we close this issue?

jcw, 2022-07-01:
Yeah, let's just close this. I admit that I don't have any ideas how this could easily be improved upon.
The option to set up a custom runner is nice. Maybe this'll bug me enough some day to have a go at it.
Many thanks for taking the time to respond.

closed 2022-07-01 by jcw