platformio/platformio-core #4342: Could not initialize project ESP32-S3

SynonZelra, 2022-07-04:
What kind of issue is this?

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Arch Linux x64 5.18.8

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.0.2

### Description of problem
can't initialize esp32-s3 project, 

#### Steps to Reproduce
New Project -> esp32-s3-devkitc -> espressif framework -> finish

### Actual Results
PIO Core Call Error: "The current working directory /mnt/Project/test project/testtest will be used for the project.\n\nThe next files/directories have been created in /mnt/Project/test project/testtest\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nResolving esp32-s3-devkitc-1 environment packages...\n\n\nError: Unknown board ID 'esp32-s3-devkitc-1'"

**The content of `platformio.ini`:**
```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = espidf
```

ivankravets, 2022-07-04:
See supported boards https://registry.platformio.org/platforms/platformio/espressif32/boards

martinberlin, 2022-11-24:
@ivankravets I really don't get why this is the accepted answer and the ticket is closed.
What  @SynonZelra describes is still reproducible today.

What is the reason that the board ID does not exist?
Is that the framework selected is espidf or has nothing to do with it?

If I select any esp32-s3-* board no matter which I still hit this **Unknown board ID**