platformio/platformio-core #4348: Feature request:  Add GitHub action workflow commands to CI output

SRGDamia1, 2022-07-09:
- [x] **Feature Request**

When looking at the results of continuous integration tests on GitHub actions, it would be *amazing* if the output could be grouped (ie, foldable) using the actions workflow commands.  This would make it much easier to find the output from the environment that failed in the log.

To do this, the output would need to include a `::group::env_name` line above the first "processing ..." line for each environment and another `::endgroup::` command at the bottom of the output for the environment.  https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#grouping-log-lines

It would also be amazing if the summary table of the ci run at the end was sent to the GitHub action summary using the $GITHUB_STEP_SUMMARY feature.  https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#adding-a-job-summary

ivankravets, 2022-07-09:
Could you try https://github.com/marketplace/actions/junit-report-action ? See https://docs.platformio.org/en/latest/core/userguide/cmd_test.html#cmdoption-pio-test-junit-output-path

Please share your experience here, we will update our docs.

SRGDamia1, 2022-07-11:
Thank you for pointing those out.  Can the `pio ci` command also output the same XML form?  I'm familiar with using GitHub actions to test for successful compilation for multiple boards, but I still haven't crossed the bridge into full unit testing.

ivankravets, 2022-07-11:
>  Can the pio ci command also output the same XML form?

No, it can't. Nevertheless, you create your own bash script that parses PlatformIO output and generates something that you need for GitHub.