platformio/platformio-core #4349: Endless scanning dependencies after update PIO

kotyara12, 2022-07-09:
Good afternoon.

Some update arrived today, probably PlatformIO 6.1.0 (judging by the time the folders were changed). After that, projects for ESP32, ESP-IDF, were no longer compiled.
The process hangs on Scanning dependencies, then there is an endless list of libraries, for half an hour now:

```
Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.40401.0 (4.4.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 51 compatible libraries
Scanning dependencies...
```

Everything was going well yesterday. Not a single project is going today. Same thing on the second computer.

ivankravets, 2022-07-09:
Could you provide a simple project to reproduce this issue? Maybe, your project but remove source code from `src` keeping "#includes". Thanks.

nodofilm, 2022-07-09:
We also experienced this issue. I unfortunately cannot provide a project as ours is private.

We've temporarily downgraded to resolve it until 6.1.1 can fix it.

Installed 6.0.2 with:

`pip install -U "platform<6.1"`

Getting the location of it with:

`which platfofmio`

Then in VSCode's Platformio Extension Settings, changing the  to a custom install with “Platformio-ide: Custom PATH”, and turning off “Platformio-ide: Use Builtin PIOCore”

VSCode raises an error when I open it. But everything compiles quickly and the dependency graph is back to normal for the time being.

ivankravets, 2022-07-09:
Sorry for the issue. Could you re-test with `pio upgrade --dev`? If it works, we will bump the 6.1.1 release.

closed 2022-07-09 by ivankravets

waghostel, 2022-07-11:
> We also experienced this issue. I unfortunately cannot provide a project as ours is private.
>
> We've temporarily downgraded to resolve it until 6.1.1 can fix it.
>
> Installed 6.0.2 with:
>
> `pip install [...]

Thank you :-D 
It works for me after install the 6.0.2 version with `pip install platformIO==6.0.2`

kotyara12, 2022-07-11:
> Sorry for the issue. Could you re-test with `pio upgrade --dev`? If it works, we will bump the 6.1.1 release.

Good afternoon

pio upgrade --dev solved the problem.
Thank you very much for the prompt help!!!

commit d24702eb (2022-07-09): Fixed an issue with endless scanning of project dependencies // Resolve #4349