platformio/platformio-core #4351: libArchive option is being ignored in 6.1.1

crespum, 2022-07-12:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Linux (Fedora 35)

**PlatformIO Version** (`platformio --version`): 6.1.1

### Description of problem
We've got a big project with many libraries under `lib` directory. Each of these libraries contains a `"libArchive": false,` option inside its corresponding `library.json`. Up to version 6.1.0, this used to work as documented: the library was not compiled into a `.a` but every source file was compiled into an object to be linked in the last build step.

Since version 6.1.1 this option seems to be ignored and the libraries are compiled to intermediate `.a` files. I suspect it has to do with [this commit](https://github.com/platformio/platformio-core/commit/31a24e1652b7633890037ac9c489e9930250b49d).

#### Steps to Reproduce

1. Clone https://github.com/crespum/arduino-blink-core-bug
2. Build with Core 6.1.0 and save the output
3. Build with Core 6.1.1 and save the output
4. Compare both outputs and see an unexpected `libtest_lib.a` in 6.1.1 ([this is the linker output taken in my setup](https://github.com/platformio/platformio-core/files/9094450/output.txt))

ivankravets, 2022-07-12:
Sorry for the issue and thanks for the report! Could you re-test with `pio upgrade --dev`? If it works for you, we will plan a hotfix release.

crespum, 2022-07-12:
Thanks so much! That seems to fix the issue. I can only compare the build results right now, but I'll test in the hardware tomorrow morning.

crespum, 2022-07-13:
I can confirm now that the firmware seems to work fine too. Again, thank you so much.

ivankravets, 2022-07-13:
Thanks for your feedback! 🙏

nerdyscout, 2022-08-02:
using v6.1.4b2 it feels like I just run into same issue defining `lib_archive = no` in platformio.ini
could it be this has been fixed only for library.json?

ivankravets, 2022-08-02:
@nerdyscout could you provide a simple PlatformIO project to reproduce your issue? Thanks!