platformio/platformio-core #4354: monitor_rts/dtr not working with ESP32 with auto-upload wiring

winnieXY, 2022-07-13:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.0

### Description of problem
I've designed a PCB with a ESP32 as controller with an automatic upload functionality using the RTS/DTR linked to the EN/IO0 of the ESP32. The upload works flawlessly and the microcontroller itself is working fine, too - however getting verbose information back via Serial communication is tricky as the monitor will use the RTS/DTR lines, too - which effectively resets the microcontroller and preventing it from running. 
I'm perfectly aware of monitor_rts/dtr = 0 and this doesn't work (I guess due to timing issues). Disabling RTS and  DTR manually within the montiro using STRG-T D and hen STRG-T R does the job however.

Is there any option available to influence the timing of these monitor_rts/dtr settings? Ideal would be a way to start with both set to "1" and then set first DTR to "0" and afterwards RTS to "0". 

See: https://community.platformio.org/t/upload-and-monitor-task-problem-with-dtr-and-rts/15947/3 for other users with the same issue. 

#### Steps to Reproduce

1. Design PCB with auto upload functionality and ESP32
[image]

2. Upload sample code (see below)
3. Attach monitor

### Actual Results
The esp is set to programming mode instead of attaching the serial

### Expected Results
The monitor is attached to the serial of the esp and you'll get serial communication back.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <arduino.h>
 void setup() {Serial.begin(115200); Serial.println("Test");  }
 void loop() { }
```

### Additional info

ivankravets, 2022-07-29:
Please reopen if you still need help

winnieXY, 2022-08-03:
> Do you mean behavior similar to https://github.com/platformio/platformio-core/blob/develop/platformio/test/runners/readers/serial.py#L52?

Yes, exactly something like this. Sorry for the delay - it's vacation time. :-)

winnieXY, 2022-08-03:
I'm not able to reopen the issue - but it would be nice if you could reopen it - the issue persists.

winnieXY, 2024-03-05:
Any news on this? There is a manual work around, however it would be cool to do this automatically