platformio/platformio-core #4355: PIO became basically unusable for debugging Marlin

EvilGremlin, 2022-07-14:
### Configuration
**Operating system**: Linux Mint 20.3
**PlatformIO Version** (`platformio --version`): 6.1.1, in VSCode

### Description of problem
Working with Marlin simulator is impossibe now, it does full rebuild every time i make a change in config (mostly enabling/disabling features) and hit Upload (to launch sim)
Nope, it's not source filters, i made it so it all stay unchanged. Not sconsign db either, i tried making it static.

#### Steps to Reproduce
1. Build Marlin's simulator_linux_release, all ok.
2. Make a change to config, say, PIDTEMPBED (yep, it's THAT bad).
3. Hit build and suffer

### Actual Results
Full rebuilt on almost every sneeze, sometimes even change of a constant triggers it.

### Expected Results
Rebuild just a few changed/affected files, like it was in somewhere in 5.x

ivankravets, 2022-07-14:
What is `PIDTEMPBED`? Yes, this is a default behavior since the first version of PlatformIO (8 years ago) - if the configuration file `platformio.ini` has been modified, the project will be rebuilt from the scratch. If you have custom configuration data that do not affect on the build process, please put them in the extra configuration files https://docs.platformio.org/en/latest/projectconf/section_platformio.html#extra-configs

cc: @thinkyhead

EvilGremlin, 2022-07-14:
@ivankravets i mean marlin's config headers, there are no changes to pio.ini or even any extra_configs

ivankravets, 2022-07-14:
If you change the file structure in the project `src` directory, it will also trigger a full rebuild. Could you provide more details on what you modified? Or, simple steps on how to reproduce this issue.

EvilGremlin, 2022-07-14:
all default bugfix-2.1, no need to even switch for sim env, just switch PIDTEMPBED in configuration.h (this just adds instance of PID, no change to include tree, no nothing) - and it'll rebuild all

EvilGremlin, 2022-07-16:
Looks like it's linux-only problem

ivankravets, 2022-07-16:
Please provide step by step guide on how to reproduce this issue, starting from cloning repo.

EvilGremlin, 2022-07-16:
ok 
1. get https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x
2. open and build (vscode in linux), all default
3. enable PIDTEMPBED in configuration.h 
4. build again, observe full rebuild

ivankravets, 2022-07-16:
Have you tried to build in the verbose mode `pio run -v`? It has a good explanation. Enabling/disabling that macro "enable/disable" extra libs/sources. What do you expect in this case? SCons tracks nested dependencies and rebuild modified nodes.

See attached logs
[a.txt](https://github.com/platformio/platformio-core/files/9125481/a.txt)
[b.txt](https://github.com/platformio/platformio-core/files/9125482/b.txt)

closed 2022-07-16 by ivankravets

EvilGremlin, 2022-07-16:
No, your logs clearly show wrong behaviour. PIDTEMPBED dont trigger any extra files, just some #ifdef. In your logs pretty much only differnce is FrameworkArduino and other side libs, and almost all Marlin files got rebuilt for no reason at all. This shouldn't happen even when there are actually extra files getting included in project, this doesn't happen in windows/macos and didn't happen in linux before.

ivankravets, 2022-07-16:
Good. Please provide PR to fix this issue.

P.S: It happens for me on macOS and it would be a bug if it does not happen.

EvilGremlin, 2022-07-16:
Hah, if only i understood all that stuff :) I peeked at pio code and decided - nope, gotta post here :) Any suggestions where to look?
 Scott said he didn't notice such thing on M1 mac. Or maybe it just built too fast...
> would be a bug if it does not happen
Why? This is clearly a major bug as it's now. Literally nothing else i've ever seen did full rebuild with only couple files chages or additional lib included or whatever.

EvilGremlin, 2022-07-16:
As far as i can see, any change to config defines, even pesky constants already in use (MAX_BED_POWER), trigger full build. Local changes in actual code, including preprocessor defines - somewhat random. *head_exploding.gif*

ivankravets, 2022-07-18:
Please forward this issue to https://github.com/MarlinFirmware/Marlin/issues

Need to ask Marlin developers if they track `Configuration.h` dynamically and change the built environment on the fly (modifying flags, source filters, etc).

I don't see issues on the PlatformIO side. You can try to create an independent PlatformIO project and try to modify a test header file. It will not recompile.

EvilGremlin, 2022-07-18:
Now i know, my expectations of compiler smarts were too high... T_T It never track actual changes, just that included header were changed, which triggers rebuild. It's just marlin is unique in that regard, two headers included EVERYWHERE, that issue getting so annoying. And sim being alreasy longest build doens't help either. There are talks now about moving away from this config style fo something like json schema.

ivankravets, 2022-07-18:
>  It's just marlin is unique in that regard, two headers included EVERYWHERE, 

Ahhh. This explains the issue. Yes, if dependent nodes are changed, the result source file will be rebuilt. Why? Because you can change any macro in a header file and the rest logic in the source file is changed as well.

See my comment here https://github.com/platformio/platformio-core/issues/4316#issuecomment-1153814893

You can change this logic in SCons or write your own decider https://scons.org/doc/production/HTML/scons-user/ch06.html#idm46358293581808