platformio/platformio-core #4363: Error when opening Serial monitor in VS Code

vabean, 2022-07-18:
Everything works fine, i can compile and upload to my D1 mini (esp8266).

But when opening Serial monitor I get this error message. I made sure that platformio is installed via 'platformio'
```
 *  Executing task in folder Blink: platformio device monitor 

Error: Traceback (most recent call last):
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/platformio/cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/user/.platformio/penv/lib/python3.10/site-packages/platformio/device/monitor/command.py", line 143, in device_monitor_cmd
    unknown_filters = set(options["filters"]) - known_filters
TypeError: 'NoneType' object is not iterable

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

 *  The terminal process "platformio 'device', 'monitor'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

gaaat98, 2022-07-18:
Can be fixed by checking that options["filters"] is not None in `/platformio/device/monitor/command.py", line 143`:
`    # check for unknown filters
    if options["filters"]:
        known_filters = set(get_available_filters())
        unknown_filters = set(options["filters"]) - known_filters

        if unknown_filters:
            options["filters"] = list(known_filters & set(options["filters"]))
            click.secho(
                ("Warning! Skipping unknown filters `%s`. Known filters are `%s`")
                % (", ".join(unknown_filters), ", ".join(sorted(known_filters))),
                fg="yellow",
            )
`

arnoldNuvisto, 2022-07-18:
Same problem for me. 

Started this morning after updating VS Code.

ivankravets, 2022-07-18:
Sorry for the issue, the quick 6.1.3 is coming soon.

commit 8c2a7df5 (2022-07-18): Fixed a regression bug when opening device monitor without any filters // Resolve #4363