platformio/platformio-core #4367: ESP-prog with esp_tool protocol fails at upload or serial

OptoStocks, 2022-07-22:
Hi guys,

I use the esp-prog as serial firmware uploading tool, with esp_tool protocol, on custom boards of mine.
Yesterday, everything was working fine about uploading and monitoring through serial. However I had to reboot my Mac today, and I can't upload or connect to serial since the reboot.

I changed cables, usb ports and esp-progs (I have 3), rebooted my Mac 2 times. Nothing helped. 
I checked with a standard dev_board connected with the same usb cable, it uploads flawlessly. 
Both VScode and platformio are up-to-date.
The esp-prog itself is still detected by platformio, as it tries to reconnect when I unplug it, and self reconnects when I plug it back (nice feature btw).

I believe that one of the very last versions of platformio got effective with the reboot of VScode, and brought some novelties I wouldn't be aware of, or maybe it's just a bug hidden somewhere.

How to solve this, and if needed, how may I help ? 
Thanks

ivankravets, 2022-07-29:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2022-07-29 by ivankravets

OptoStocks, 2022-07-29:
Hi Ivan,

I just made the test ! 
Well done, it connects and programs now.
Serial connection and monitoring are working too.

Thank you for your work and time, and thank you for offering the opportunity to give a hand in the process.

@Sinclair732, sorry for not answering earlier : you seem to face a different problem, here was a mis-selection of the port of the Program Interface (applies to esp-prog only).

ivankravets, 2022-07-30:
Great! Happy coding with PlatformIO! 🚀

commit 7c9e0393 (2022-07-29): Improved device port finder when using dual channel UART converter // Resolve #4367