platformio/platformio-core #4370: PIO Core >6.1.0 new LDF including libraries that shouldn't be included.

Timvrakas, 2022-07-25:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: GH Actions CI, and MacOS AArch64.

**PlatformIO Version** (`platformio --version`): >6.1.0

### Description of problem
the changes introduced in 6.1.0 are causing strange behavior. 

I have a library (called `SSIEth`) that lives in `lib/SSIEth`. For some build targets, it needs to be included, and for other, it needs to not be included. In the past, the LDF (in chain mode) would evaluate the preprocessor macros, and include the library only when needed. Now, it's always being included. I bisected PIO-core versions and this issue appears at 6.1.0, but not at 6.0.2.

I have been able to resolve the issue with a `lib_ignore = SSIEth` line in the platformio.ini, but as I understand it, this behavior is not correct, so I wanted to report the issue.

#### Steps to Reproduce

The issue can be inspected by looking at the CI runs here:
https://github.com/stanford-ssi/SpaceSalmon

This commit fails (unrelated to the contents of the commit): https://github.com/stanford-ssi/SpaceSalmon/commit/2b907282cdfdb23842664ab62af747b0c3612346
and the next commit succeeds: https://github.com/stanford-ssi/SpaceSalmon/commit/b1bde69ca3cc820ea7552f56bb238c19adc0646c

### Actual Results
Without the `lib_ignore = SSIEth` line, the new LDF tries to include the unused library, causes a build failure.

### Expected Results
I would expect that libraries in the `lib/` directory that are not used by a given build configuration should not be included in the build. This is the behavior on pio-core 6.0.2

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
https://github.com/stanford-ssi/SpaceSalmon/blob/b1bde69ca3cc820ea7552f56bb238c19adc0646c/platformio.ini

ivankravets, 2022-07-25:
It was a bug in PlatformIO Core < 6.1. The current behavior is valid. 

> In the past, the LDF (in chain mode) would evaluate the preprocessor macros, 

See https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

You use an incorrect value for [lib_ldf_mode](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#projectconf-lib-ldf-mode), it should be `chain+`.

Timvrakas, 2022-07-26:
Thanks, that worked!

MrKevinWeiss, 2022-08-01:
Hmm I am noticing a similar issue.  It seems the LDF does not properly resolve when I use an `ubuntu-20.04` GH actions box with `pip install platformio==6.1.3`.  On my local windows box it can figure it out and the
```c
#define IGNORE_THIS_LIBRARY
#if IGNORE_THIS_LIBRARY
#include "this_library.h"
#endif
```

prevents the libraries from getting included (as expected with `lib_ldf_mode = chain+`).

Can anyone confirm this still functions with ubuntu?