platformio/platformio-core #4371: undefinedError: Program/firmware is missed

dronecz, 2022-07-25:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64 

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.4a2

### Description of problem

When I try to debug Marlin fw, I can build image, but when I hit `Start debugging (F5)` I get following error: 

[image]

I can start and debug other files (arduino-blink for ESP32) from PIO examples.  

#### Steps to Reproduce

1. [Download ](https://marlinfw.org/meta/download/)Marlin fw
[image]
3. Extract content
4. Replace files from `files_to.reproduce_error.zip` file (just CTRL+A -> CTRL+C and CTRL+V in Marlin-2.1.x folder)
5.  Build firmware 
[image]
6. Hit `Start debugging (F5)` and look for the result in `DEBUG CONSOLE`

### Actual Results
Unable to use debbuger with Marlin fw. 

### Expected Results

Able to use debbuger with Marlin fw. 
### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
#
# Marlin Firmware
# PlatformIO Configuration File
#
# For detailed documentation with EXAMPLES:
#
# https://docs.platformio.org/en/latest/projectconf/index.html
#

# Automatic targets - enable auto-uploading
#targets = upload

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = esp32
include_dir  = Marlin
extra_configs =
    ini/avr.ini
    ini/due.ini
    ini/esp32.ini
    ini/features.ini
    ini/lpc176x.ini
    ini/native.ini
    ini/samd51.ini
    ini/stm32-common.ini
    ini/stm32f0.ini
    ini/stm32f1-maple.ini
    ini/stm32f1.ini
[... 246 lines cut ...]
monitor_flags     =
  --quiet
  --echo
  --eol
    LF
  --filter
    colorize
  --filter
    time

#
# Just print the dependency tree
#
[env:include_tree]
platform         = atmelavr
board            = megaatmega2560
build_flags      = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
build_src_filter = +<src/MarlinCore.cpp>
```

**Source file to reproduce issue:**
[files_to reproduce_error.zip](https://github.com/platformio/platformio-core/files/9180871/files_to.reproduce_error.zip)

### Additional info
None

ivankravets, 2022-07-25:
Have you reported this issue to the https://github.com/MarlinFirmware/Marlin/issues firstly?

dronecz, 2022-07-25:
No, but I discussed this error with devs at discord and nobody had similar issue.

I was instructed [in the forum](https://community.platformio.org/t/undefinederror-program-firmware-is-missed/28921) to create issue here. 

I tried to find similar error on the Internet and Github but without luck, except source code of PIO core. PIO does not provide enough info about this error so I do not know, if this is related to some path in Marlin or PIO in general.

ivankravets, 2022-07-30:
This is an issue of Marlin. See the solution in https://github.com/MarlinFirmware/Marlin/pull/24569.

closed 2022-07-30 by ivankravets