platformio/platformio-core #4375: Add support for Spresense

EguchiTomoaki, 2022-08-01:
Please add support for Spresense

Document : https://developer.sony.com/develop/spresense/docs/home_en.html

maxgerhardt, 2022-08-01:
Are you more interested in the [Arduino core](https://github.com/sonydevworld/spresense-arduino-compatible) or Spresense SDK (aka, NuttX)?

EguchiTomoaki, 2022-08-01:
I'm interested in the Arduino core.

maxgerhardt, 2022-08-05:
@EguchiTomoaki The platform is ready for its first test with a real board, but I don't have one. Can you follow the instructions at https://github.com/maxgerhardt/platform-sonyspresense and see if `arduino-blinky` works?

EguchiTomoaki, 2022-08-06:
`arduino-blinky` worked on all cores.
Also, when I tried changing the `LED_BUILTIN` on line 3 to `LED1`, LED1 blinked normally. *(1)
I will try other sample programs from now.

↓ (1) Blinking of LED1

https://user-images.githubusercontent.com/93034341/183250721-9ba952d0-3d01-4f0e-8a45-b279be0ec19e.MOV

maxgerhardt, 2022-08-06:
That's great! 

Does the `Serial` output from the arduino-blinky example also appear when using the "monitor" task?

And in [the examples](https://github.com/maxgerhardt/platform-sonyspresense/tree/main/examples) do the GNSS and multiprocessing-message examples work too? (Latter one has special instructions for flashing).

EguchiTomoaki, 2022-08-07:
I confirmed that `arduino-blinky` outputs serial every second as shown in the picture.

[image]

I also confirmed that `GNSS` successfully obtains location information from GPS.

And I confirmed that `multiprocessing-message` can communicate with the main core and all sub-cores as shown in the video below.

https://user-images.githubusercontent.com/93034341/183296184-a1484d15-ff32-4a16-aaf5-60efc1528dc5.mov

These three examples are working as well as they did when run on the Arduino ide so far.

maxgerhardt, 2022-08-09:
Great, I'll add the MP3DEC downloader and MP3Player sample in a bit for you to test, but it seems it all works properly and is equal with the Arduino IDE featureset.

Tanmax-Payne, 2024-03-02:
Thanks devs for supporting this not-so-mainstream MCU, is there any usable progress in the main for the main branch?

maxgerhardt, 2024-03-03:
I've updated the project to use the latest SPresense Arduino core v3.2.0 and added more examples. (Update [per instructions](https://github.com/maxgerhardt/platform-sonyspresense?tab=readme-ov-file#updating)).