platformio/platformio-core #4379: Build failes on ESP32 in Pyhton Code with Version 4.3.0

savejeff, 2022-08-04:
Hi, 

since this moring (with no code changes) the build of my project failed in the python code. here is the traceback

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
KeyError: 'IDE_EXTRA_DATA':
  File "C:\Users\AUSER\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 188:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\AUSER\.platformio\platforms\espressif32\builder\main.py", line 287:
    target_elf = env.BuildProgram()
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "C:\Users\AUSER\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "C:\Users\AUSER\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "C:\Users\AUSER\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 326:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 660:
    return method(*args, **kw)
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\AUSER\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 44:
    env["IDE_EXTRA_DATA"].update({"application_offset": env.subst("$ESP32_APP_OFFSET")})
  File "C:\Users\AUSER\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Environment.py", line 387:
    return self._dict[key]
```

This happens with ESP32 only as far as I can see. RP2040 compiles with no problem.

here is my platformio.ini config
```

[env:BOARD_ESP32]
platform = espressif32@4.3.0
board = esp32dev

```
this seems to not happen with platform = espressif32@5.1.0 
i think this is a backward compatibility problem.

ivankravets, 2022-08-04:
Thanks for the bug report. Please re-test with `pio upgrade --dev`.

closed 2022-08-04 by ivankravets

savejeff, 2022-08-04:
Works now. Thx!

commit 59902abd (2022-08-04): Keep support for legacy "IDE_EXTRA_DATA" // Resolve #4379