platformio/platformio-core #4381: Can I disable automatic update of PIO?

solawc, 2022-08-05:
I use platform IO to do my engineering projects. It is very easy to use. It can manage multiple projects in a unified way. These functions are great! Recently, my colleagues have encountered some problems. I also know that PIO is updated frequently, which is a good thing in itself. But as a developer, if the current version is stable and available, and the project has not been completed, suddenly update the compilation tool, which will obviously lead to some unstable factors. After the PIO on my colleagues' computer is automatically updated, it can not be compiled normally, and the task is always loaded, It can be solved after reinstallation, but this is obviously not what I want. If he can disable automatic update, I will ask my colleagues or friends to develop with the same version number until the project is completed and the update is evaluated.

ivankravets, 2022-08-05:
This is not a problem of PlatformIO Core. You need to use a declarative approach based on semantic versioning. See:

- https://docs.platformio.org/en/latest/core/migration.html#migration-steps
- https://docs.platformio.org/en/latest/librarymanager/dependencies.html

Could you share your `platformio.ini` file?

solawc, 2022-08-07:
> This is not a problem of PlatformIO Core. You need to use a declarative approach based on semantic versioning. See:
>
> * https://docs.platformio.org/en/latest/core/migration.html#migration-steps
> * [...]

https://github.com/solawc/Open_GRBL/blob/main/platformio.ini

My configuration is almost the same as it, but there is a difference in - D, that is, some macro definitions. It is OK on my PC. after the update a few days ago, there is a problem on my colleague's computer. Win10 is used, and PIO is run on vscode.

ivankravets, 2022-08-08:
> after the update a few days ago, there is a problem on my colleague's computer. Win10 is used, and PIO is run on vscode.

Could you provide more details about the problems of your colleague? Your `platformio.ini` looks good, you added semantic requirements.

closed 2022-08-15 by ivankravets