platformio/platformio-core #4382: Googletest in a linux arm64 environment failing

BadenMorgan, 2022-08-05:
### Configuration

**Operating system**:
Linux ARM64

**PlatformIO Version** (`platformio --version`):

### Description of problem

Googletests keep failing before compilation completes with an error:
`xtensa-esp32-elf-g++: error: unrecognized command line option '-pthread'`
I can run them on a Windows and Mac environment and have notices, atleast on windows that the parameters -pthread is not used but instead -lpthread. This was found by putting the pio test function into -vvv. I can post partial logs only as -vvv's output is very long it would be unreasonable to post a log with 10s of thousands of lines.

#### Steps to Reproduce

1. pio test -vv "<path to ini file>" --upload-port /dev/ttyUSB0 --test-port /dev/ttyUSB0 -d <path to project>
2. wait

### Actual Results

```
Linking .pio/build/esp32dev/firmware.elf
xtensa-esp32-elf-g++: error: unrecognized command line option '-pthread'
*** [.pio/build/esp32dev/firmware.elf] Error 1
Uploading stage has failed, see errors above. Use `pio test -vvv` option to enable verbose output.
------------ esp32dev:test_file2 [ERRORED] Took 44.38 seconds ------------
=================================== SUMMARY ===================================

Environment    Test                Status    Duration
-------------  ------------------  --------  ------------
esp32dev       test_unity_tests       SKIPPED
esp32dev       test_file1  ERRORED   00:02:52.913
esp32dev       test_file2     ERRORED   00:00:44.385
================== 2 test cases: 0 succeeded in 00:03:37.298 ==================
```

### Expected Results

```
------------- esp32dev:test_file2 [PASSED] Took 19.18 seconds -------------

===================================== SUMMARY ===================================== 
Environment    Test                Status    Duration
-------------  ------------------  --------  ------------
esp32dev       test_unity_tests       SKIPPED
esp32dev       test_file1     PASSED    00:00:27.853
esp32dev       test_file2     PASSED    00:00:19.182
=================== 13 test cases: 13 succeeded in 00:00:47.034 =================== 
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
Not an exact copy of the ini file as there are some proprietary includes but it's just the libdeps I left out. We are using an older platform version because the code does not compile on the newer versions and we just haven't had the time to get it to work with newer versions. But I don't believe it is the platform version as the project builds in windows and mac with the exact same configuration.
```ini
[env:esp-wrover-kit]
platform = espressif32@3.5
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_port = <port>
monitor_port = <port>
test_port = <port>
targets = upload, monitor
monitor_filters = time, esp32_exception_decoder, printable
build_flags = -DCORE_DEBUG_LEVEL=5
test_framework = googletest
```

**Source file to reproduce issue:**
Don't believe the issue isn't related to source code, have run it on multiple projects with different source code but similar ini files and environments

ivankravets, 2022-08-09:
Could open `.pio/libdeps/esp-wrover-kit/googletest/platformio-build.py` file and replace (comment) the last line with 
```python
    # globalenv.Append(LINKFLAGS=["-pthread"])
```

Does it work now?

BadenMorgan, 2022-08-10:
Forgot to add that I was using PIO v6.1.3

Applied the change you recommended and it worked, thank you.

Is the platformio-build.py file added by PIO when it downloads the library?

ivankravets, 2022-08-12:
Please run the following commands in your project:

```
pio system prune
pio run -t cleanall
```

It should work now.

closed 2022-08-12 by ivankravets