platformio/platformio-core #4383: About pio core override extensions.json enhancement

seishinkouki, 2022-08-05:
- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

Currently, pio will keep user custom [recommendations] section when overriding extensions.json, I think it will be helpful keep both custom [recommendations] and [unwanted Recommendations] sections.

ivankravets, 2022-08-09:
Please re-test with `pio upgrade --dev`.

closed 2022-08-09 by ivankravets

seishinkouki, 2022-08-09:
> Please re-test with `pio upgrade --dev`.

it work as expected, thanks

commit bc9d9ac2 (2022-08-09): Keep custom "unwantedRecommendations" when generating projects for VSCode // Resolve #4383