platformio/platformio-core #4384: Incorrect verbose command-line output

daniel-starke, 2022-08-05:
What kind of issue is this?

- [x] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 7 Pro SP1

**PlatformIO Version** (`platformio --version`): 6.1.3

### Description of problem

Command-line set pre-processor defines did show with `pio run -v` but were not passed to the compiler.

#### Steps to Reproduce

1. Set pre-processor defines for `env` and `DefaultEnvironment()` but not `projenv`.
2. Compile with `pio run -v` and check for the define within the library code.

### Actual Results

The verbose output with the called command-line shows the `-D` argument. However, the compiler complains that it was not given.

### Expected Results

The verbose output of the called command-line shows the `-D` argument not which reflects the actually called command-line.  
There exists a discrepancy between the verbose output and the actual command-line.

daniel-starke, 2022-08-05:
Note that this behavior changed from PlatformIO 5 to 6.

ivankravets, 2022-08-08:
Please provide a simple PlatformIO project to reproduce this issue.

daniel-starke, 2022-08-08:
https://github.com/daniel-starke/STM32CubeDuino/tree/406c190f7d5021062c212afee2d1eef39fc0af6d fails and https://github.com/daniel-starke/STM32CubeDuino/tree/795503d2a87efb6e4c26dae5cca0715583a3be99 works. This is essentially due to [this](https://github.com/daniel-starke/STM32CubeDuino/blob/795503d2a87efb6e4c26dae5cca0715583a3be99/BuildScript.py#L1678) added line.
If you run the test cases like in [the Github action](https://github.com/daniel-starke/STM32CubeDuino/blob/795503d2a87efb6e4c26dae5cca0715583a3be99/.github/workflows/compatibility-tests.yml#L87-L94) with `-v` added to the `pio` call you will notice no change in the called command-line, comparing both versions of `BuildScript.py`. However, the missing `projenv` configuration causes missing command-line parameters.

    error: #error "Please select first the target STM32F1xx device used in your application (in stm32f1xx.h file)"

The define was passed according to the verbose output but is missing at compile-time. The only obvious reason for this is an error in the verbose PlatformIO output.

ivankravets, 2022-08-09:
Sorry, I didn't dip into your codebase. It seems you try to include some files from project source files (projenv), and you want to expect flags set by "libenv". This impossible. So, when PlatformIO builds the library and handles includes, it uses "libenv", however, when you include external files in your projects `src` files - this is a typical GCC replacement.

Did this hint help you?

daniel-starke, 2022-08-09:
Don't get me wrong. I found a solution for my specific issue.
What I claim is that the flags actually passed to the compiler are not reflected by the verbose command-line output of PlatformIO.

ivankravets, 2022-08-09:
Could you provide a separate simple PlatformIO-project to reproduce this issue?

daniel-starke, 2022-08-09:
I have tried to reproduce the issue and found out that it was caused due to interleaved outputs from multiple jobs.
I did not consider that multiple build jobs are involved and the verbose command-line output may not correspond to the compiler output. That is why I could not relate the compiler output to the printed command-line.
Calling `pio` with `-j 1` gives me a proper output.

ivankravets, 2022-08-09:
Ahh, great! Happy to hear that you solved this issue. Could I close this ticket?

daniel-starke, 2022-08-09:
Yes, please.

closed 2022-08-09 by ivankravets