platformio/platformio-core #4385: BuildSources() not work from post : script.py

Wiz-IO, 2022-08-06:
I want to add and compile code from not compiled framework folder
but **BuildSources() not work from POST:SCRIPT.PY**

as example:
https://github.com/Wiz-IO/TEST/blob/main/PICO-LWIP-APP/extra_script.py

pre:scrypt.py
BuildSources() work
BuildLibrary() work

**post:scrypt.py**
BuildLibrary() work
**BuildSources() not work … Is it a BUG?** or I make any wrong...

ivankravets, 2022-08-08:
Hi Georgi,

This is a feature, not a bug. The inline source files are handled at the [BuildProgram](https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/piobuild.py#L59) step which you call in your platformio/main.py 

So, any calling of `BuildSources`later does not have any effect. We will inform the user about this issue.

Wiz-IO, 2022-08-08:
Hi Ivan, thanks for the answer

my thought was: add_module( src folder ) from ini file ... CMAKE uses similar practices
some code is not compiled by main builder, but exist in framework
example: 
lwip is compiled as basic, without apps ... http, tftp etc
and I wanted to add them from INI with POST script
for now I found a way with PRE script - I had uninitialized variables

```
...
Building in release mode
-- EXECUTE POST-SCRIPT.PY --
-- FROM PRE-SCRIPT - BuildSources() -- Compiling WORK
-- FROM MAIN BUILDER --
Compiling .pio\build\...any code...
Compiling .pio\build\any-framework\main.o
-- FROM POST-SCRIPT - BuildSources() -- Would be great feature <<<----------
-- FROM POST-SCRIPT - BuildLibrary() -- Compiling WORK
Linking .pio\build\raspberry-pi-pico\APPLICATION.elf
...
```

you can close issue if not important

ivankravets, 2022-08-08:
I've just added a warning message. Please use BuildLibrary in this case.

closed 2022-08-08 by ivankravets

Wiz-IO, 2022-08-08:
BTW in doc
https://docs.platformio.org/en/latest/scripting/actions.html

need more info about action target( trigers )
env.AddPreAction(TRIGER", before)
env.AddPostAction(AFTER, after)

maybe exist: before_main_build, after_main_build

ivankravets, 2022-08-08:
Pre/Post action accepts a taget (or list of targets) as a first argument. Target can be a physical path, a SCons node, or virtual alias.

Wiz-IO, 2022-08-08:
I understood

Wiz-IO, 2022-08-13:
as idea / info

I make experiment with add "modules" ( not compiled codes, existing in framework ) - **makefile style**

as example:
https://github.com/Wiz-IO/framework-wizio-pico/tree/main/SDK/lib/at_tok

**at_tok** is not compiled from my builder by default and I add this module with project ini
board_build.modules = $FRAMEWORK_DIR/SDK/lib/at_tok

the "module" can be existing code in framework or
can be internal or external library ... lib_deps etc
can be compiled "on request"

my add modules function ... 
https://github.com/Wiz-IO/wizio-pico/blob/e2ca0e87724e0fd8cd90b09563aa3aea5e6af232/builder/frameworks/common.py#L348-L364

commit dfffd5e9 (2022-08-08): Warn about calling “env.BuildSources” in a POST-type script // Resolve #4385