platformio/platformio-core #4386: Confusing behaviour for Clean All

DigiH, 2022-08-09:
With recent updates of Visual Studio Code (1.17.0) and Platformio (Core 6.1.3/Home 3.4.3/IDEv2.5.1) on macOS 10.15.7 I'm experiencing strange behaviours when trying to select the **Clean All** project task in an environment.

**Clean** works fine as usual, cleaning the build directory of the environment.

**Clean All** however, which previously nicely also cleaned the libdeps of the project environment, now only brings up the task menu, with the shown three options when clicking on the entered Clean All command

[image]

[image]

Also when using the cli command for cleanall, e.g.

`platformio run --target cleanall --environment esp32dev-ble`

it cleans ALL libdeps for all previously built environments, but only the build directory for the specified environment.

I'm pretty confident that when the actual **Clean All** project task was still working for me previously it only ever cleared/deleted both the build directory and the libdeps only for the environment it was selected in.

Thought I had might have something messed up on my system, but just confirmed with a colleague confirming the same behaviour on his macOS 12.5 and all the latest relevant versions as myself.

closed 2022-08-10 by ivankravets

DigiH, 2022-08-10:
Hi @ivankravets , thanks a lot for the very quick fix, which seemed to have solved the cli command issue mentioned above

`platformio run --target cleanall --environment esp32dev-ble`

in the --dev upgrade. 👍 

The sidebar task **Clean All** still is broken though. 

Searching for **Clean** tasks in the Platformio tasks brings up 

[image]

with the `PlatformIO: Clean a build environment and installed library dependencies` being the `cleanall` task, but not named/linked to the **Clean All** sidebar project task any longer.

Or would this be addressable in [platformio-vscode-ide](https://github.com/platformio/platformio-vscode-ide) ?

Thanks

ivankravets, 2022-08-10:
It looks like an issue in VSCode, see https://github.com/microsoft/vscode/issues/157814

Could you file a new issue in https://github.com/platformio/platformio-vscode-ide/issues ? We will make a hotfix to it and release a new version of extension. Thanks!

commit 847fdd4d (2022-08-10): Fixed an issue when the "cleanall" target removes dependencies from all working environments // Resolve #4386