platformio/platformio-core #4387: Request support for board uStepper S based stepper driver and encoder

smukkejohan, 2022-08-10:
Please add support for uStepper S made by ON Development (https://www.ustepper.com/) 

uStepper S is an Arduino compatible board, with integrated stepper driver and 16-bit rotary encoder, mounted directly on the back of a NEMA 17 size stepper motor.

- Uses a 328PB MCU
- Installation instructions for use with Arduino IDE here as well as all relevant src code https://github.com/uStepper/uStepperS

maxgerhardt, 2022-08-10:
Can be made to work already by adding the variant folder and board definition to the project: See [forum](https://community.platformio.org/t/add-ustepper-s-board/26154/7) and [project](https://github.com/maxgerhardt/pio-usteppers).

ivankravets, 2022-08-11:
Thanks, @maxgerhardt, for the provided solution!

maxgerhardt, 2022-08-11:
This however slightly outdated, the core has moved on, and has also modified `libraries/`, which now would need a new core package to integrate it correctly, instead of my solution of just copying the variant folder (+ new board JSON).