platformio/platformio-core #4389: PIO Registry Accepts invalid SemVers

maxgerhardt, 2022-08-10:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`):  6.1.4b4

### Description of problem

https://registry.platformio.org/libraries/nanopb/Nanopb/versions has library versions that do not conform to the syntax X.Y.Z, yet are listed in the registry. PlatformIO crashes upon trying to download them, although the exact command specified in the installation tab is executed.

#### Steps to Reproduce

1. Open a PIO CLI
2. Execute `pio pkg install -g -l "nanopb/Nanopb@^0.4.6.4"`

### Actual Results

```
>pio pkg install -g -l "nanopb/Nanopb@^0.4.6.4"
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 6.1.4b2!

************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
************************************************************************************************************************
Error: Traceback (most recent call last):
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 65, in package_install_cmd
    install_global_dependencies(options)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\commands\install.py", line 89, in install_global_dependencies
    lm.install(
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 62, in _install
    spec = self.ensure_spec(spec)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\manager\base.py", line 133, in ensure_spec
    return spec if isinstance(spec, PackageSpec) else PackageSpec(spec)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 184, in __init__
    self._parse(self.raw)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 291, in _parse
    raw = parser(raw)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 316, in _parse_requirements
    self.requirements = tokens[1].strip()
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\platformio\package\meta.py", line 231, in requirements
    else semantic_version.SimpleSpec(str(value))
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\Max Gerhardt\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block '^0.4.6.4'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

### Expected Results

Installation instructions always work, PIO registry rejects invalid SemVers.

### If problems with PlatformIO Build System:

-/-

### Additional info

Conforming versions work fine.

```
>pio pkg install -g -l "nanopb/Nanopb@0.4.6"
Library Manager: Installing nanopb/Nanopb @ 0.4.6
Unpacking  [####################################]  100%
Library Manager: Nanopb@0.4.6+4 has been installed!
```

ivankravets, 2022-08-11:
Thanks, @maxgerhardt! We will hide specifications for the non-semantic versions.

@PetteriAimonen, do you plan to back to the semantic versioning? https://semver.org/

PetteriAimonen, 2022-08-11:
@ivankravets Eventually, yes. But I'm not going to change my versioning just because of one distribution platform. Even though platform.io is a nice platform for users, it's been giving me pretty much trouble as a publisher and I'm reconsidering whether it is even worth it.

ivankravets, 2022-08-11:
@PetteriAimonen, sorry for the issue. If you manage library updates to the PlatformIO Registry manually, PlatformIO Registry's Crawler will not pull any versions automatically. Would you like me to disable automatic updates for your package?

If you have a new version, you will need to publish it manually using `pio pkg publish --owner nanopb` command. This is the recommended way.

Currently, the legacy crawler monitors this line https://github.com/nanopb/nanopb/blob/master/library.json#L3 . Once it changes, it will treat the state of that commit as a new release.

PetteriAimonen, 2022-08-11:
@ivankravets No, please do not. Even as is, I'm extremely short on time I can spend on nanopb, and any extra changes such as trying to rework versioning and doing manual publish etc. just increases the impossibility.

ivankravets, 2022-08-11:
> In my opinion if platformio cannot work with 0.4.6.4, people can just go and download the package manually. 

They can work, but they can't use it for decorative embedded development. Only legacy workflow which specifying the direct version of the library. The issue arises why someone wants to lock the PlatformIO-project to the MINOR version using `^` sign, for example, `^0.4.6`.

ivankravets, 2023-04-21:
PIO Registry does not accept packages if the `version` does not comply with https://semver.org/

closed 2023-04-21 by ivankravets