platformio/platformio-core #4390: PIO Home not showing up

TheMakerJojo, 2022-08-10:
Hello.
I have a problem using PlatformIO in codeserver. 
I installed the extension using the VSIX-File from the VS code marketplace.
Codeserver is running on an Raspberry Pi 4b with 2GB of RAM which is connected to my iPad Pro M1 via usb-c cable.
The connection is set up like in this tutorial: https://www.hardill.me.uk/wordpress/2019/11/02/pi4-usb-c-gadget/

The PIO Home won’t show up in codeserver. See in the picture below:
[image]

If i do „pio home --host 10.55.0.1“ to display PIO Home in my Webbrowser it works perfectly fine.
[image]
[image]

Please help me!

ivankravets, 2022-08-12:
Please take a look at the [platformio-ide.pioHomeServerHttpHost](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-piohomeserverhttphost) setting.

closed 2022-08-12 by ivankravets

TheMakerJojo, 2022-08-12:
Thanks for the answer.

I tried 0.0.0.0, 127.0.0.1 and 10.55.0.1 (IP-Address of the Raspberry Pi), but it didn't work.

Does the Raspberry Pi (vs codeserver) need an internet connection?