platformio/platformio-core #4393: Platformio CI ignores ARDUINO_ARCH_ macro's in .ino

Erriez, 2022-08-13:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Not relevant (Ubuntu 22.04)

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.4

### Description of problem

Sketch contains `ARDUINO_ARCH_` macro's to define pins for different architectures:

**examples/Test/Test.ino**
```c++
#include <Arduino.h>

#if defined(ARDUINO_ARCH_AVR) // <- Macro ignored by platform ci
#include <LowPower.h> // AVR library included incorrectly by platform ci and is not designed for ESP8266/ESP32 -> Build error
#define TX_PIN           D2
#elif defined(ARDUINO_ARCH_ESP8266)
#define TX_PIN           4
#elif defined(ARDUINO_ARCH_ESP32)
#define TX_PIN           21
#else
#error "May work, but not tested"
#endif

void setup()
{
}

void loop()
{
}
```

Building for ESP8266 shall not include `LowPower.h`, but the build macro's `ARDUINO_ARCH_` are ignored: 

```bash
$ pio pkg install --global --library "Low-Power"
$ platformio ci -v --board d1_mini examples/Test/Test.ino

Output:
...
xtensa-lx106-elf-g++ -o "/tmp/tmph9x34b7g/src/Test.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpfnty_2w6"
/tmp/tmph9x34b7g/src/Test.ino:7: warning: "RF_TX_PIN" redefined
    7 | #define RF_TX_PIN           4
      | 
/tmp/tmph9x34b7g/src/Test.ino:5: note: this is the location of the previous definition
    5 | #define RF_TX_PIN           3
      | 
/tmp/tmph9x34b7g/src/Test.ino:9: warning: "RF_TX_PIN" redefined
    9 | #define RF_TX_PIN           1
      | 
/tmp/tmph9x34b7g/src/Test.ino:7: note: this is the location of the previous definition
    7 | #define RF_TX_PIN           4
      | 
...
In file included from /home/a/.platformio/lib/Low-Power/LowPower.cpp:49:
/home/a/.platformio/lib/Low-Power/LowPower.h:167:5: error: #error "Processor architecture is not supported."
  167 |    #error "Processor architecture is not supported."
      |     ^~~~~
/home/a/.platformio/lib/Low-Power/LowPower.cpp:1179:3: error: #error "Processor architecture is not supported."
 1179 |  #error "Processor architecture is not supported."
      |   ^~~~~
...
*** [.pio/build/d1_mini/lib372/Low-Power/LowPower.cpp.o] Error 1
```

#### Steps to Reproduce

See previous:
* Sketch compiles correctly with Arduino IDE for UNO / Generic ESP8266 or ESP32.
* Building the `Test.ino` sketch with `platformio ci` does not work for non AVR.

### Expected Results

Successful build.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
Not available.

**Source file to reproduce issue:**
See previous.

### Additional info

Macro `ARDUINO_ARCH_ESP8266` is configured correctly when building `Test.ino`:

```bash
xtensa-lx106-elf-g++ -o .pio/build/d1_mini/src/Test.ino.cpp.o -c -fno-rtti -std=gnu++17 -fno-exceptions -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -D_GNU_SOURCE -ffunction-sections -fdata-sections -Wall -Werror=return-type -free -fipa-pta -DPLATFORMIO=60104 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_WEMOS_D1MINI -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_D1_MINI\" -DFLASHMODE_DIO -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -Iinclude -Isrc -I/home/a/.platformio/lib/Low-Power -I/home/a/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/home/a/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/home/a/.platformio/packages/toolchain-xtensa/include -I/home/a/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include -I/home/a/.platformio/packages/framework-arduinoespressif8266/variants/d1_mini src/Test.ino.cpp
```

Any suggestions?

maxgerhardt, 2022-08-15:
As [documented](https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode), if you use conditional `#ifdef` compilation, to include libraries, you need to set the `lib_ldf_mode` to `chain+` or `deep+`. As [documented](https://docs.platformio.org/en/latest/core/userguide/cmd_ci.html#cmdoption-pio-ci-O, you can add `-O "lib_ldf_mode=deep+` to the `pio ci` command to achieve this.

ivankravets, 2022-08-15:
Could you try this option https://docs.platformio.org/en/latest/core/userguide/cmd_ci.html#cmdoption-pio-ci-O?

```
platformio ci -O "lib_ldf_mode=chain+" --board d1_mini examples/Test/Test.ino
```

Erriez, 2022-08-15:
Thanks for your fast reply, @ivankravets. It solved the build error:

```bash
$ platformio ci -O "lib_ldf_mode=chain+" --board uno --board d1_mini examples/Test/Test.ino

Output:
...
Converting Test.ino
/tmp/tmp1zx12mql/src/Test.ino:7: warning: "TX_PIN" redefined
    7 | #define TX_PIN           4
      | 
/tmp/tmp1zx12mql/src/Test.ino:5: note: this is the location of the previous definition
    5 | #define TX_PIN           D2
      | 
/tmp/tmp1zx12mql/src/Test.ino:9: warning: "TX_PIN" redefined
    9 | #define TX_PIN           21
      | 
/tmp/tmp1zx12mql/src/Test.ino:7: note: this is the location of the previous definition
    7 | #define TX_PIN           4
      | 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
...
Environment    Status    Duration
-------------  --------  ------------
uno            SUCCESS   00:00:00.460
d1_mini        SUCCESS   00:00:04.714
```

It still reports the redefined "TX_PIN" warning, but handles the `LowPower.h` include for AVR targets correctly.

ivankravets, 2022-08-15:
Great! Happy coding with PlatformIO! 🚀

closed 2022-08-15 by ivankravets