platformio/platformio-core #4396: & (ampersand) in a build flag results in environment interpreter error

tartinesKiller, 2022-08-16:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 11, Ubuntu 22.04
**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.4
### Description of problem
I'm trying to override the default font in lvgl, which is done through a define, `CONFIG_LV_FONT_DEFAULT`, normally located in a configuration file header. It is done by setting this define to the address of a font (e.g. `#define CONFIG_LV_FONT_DEFAULT &roboto14`).
However, with platformio, one must pass this define through the build_flags, like so:
` -D CONFIG_LV_FONT_DEFAULT="&roboto14"`
However, every time there is an ampersand in one of the value of build_flags, it seems the value (without the &) is evaluated by the OS interpreter.

#### Steps to Reproduce

1. Clone https://github.com/tartinesKiller/lv_platformio_buildenv_bug
2. Try to build
3. In order ro rule out a bug in lvgl, I define `TEST_AMP` as `&a`, `a` being an int. If you want to test against lvgl, please comment line 25 of platformio.ini, as well as lines 20-21 of main.c

### Actual Results
Build fails, with the following build log:
```
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 8.2.0
|-- lv_drivers @ 8.2.0
|   |-- lvgl @ 8.2.0
Building in release mode
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_disp.o
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_event.o
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_group.o
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_indev.o
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_indev_scroll.o
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_obj.o
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_obj_class.o
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\emulator_64bits\liba06\lvgl\core\lv_obj_draw.o
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_disp.o] Error 1
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_event.o] Error 1
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_group.o] Error 1
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_indev.o] Error 1
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_indev_scroll.o] Error 1
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_obj.o] Error 1
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_obj_class.o] Error 1
gcc: fatal error: no input files
compilation terminated.
'a' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\emulator_64bits\liba06\lvgl\core\lv_obj_draw.o] Error 1
```
We can see several lines `'a' is not recognized as an internal or external command, operable program or batch file.`, which lead to think pio try to execute the value of the faulty define against the OS interpreter. 
On Ubuntu, same behaviour, with: `sh: 1: a: not found`

### Expected Results
Should build successfully, with TEST_AMP correctly interpreted. 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = emulator_64bits

; Shared options
[env]
build_flags =
  ; Don't use lv_conf.h. Tweak params via platfom.ini.
  -D LV_CONF_SKIP
  -D LV_CONF_INCLUDE_SIMPLE
  ; Enable LVGL demo, remove when working on your own project
  -D LV_USE_DEMO_WIDGETS=1
  ; Add more defines below to overide lvgl:/src/lv_conf_simple.h
  -D LV_FONT_CUSTOM_DECLARE="LV_FONT_DECLARE(roboto10) LV_FONT_DECLARE(roboto14) LV_FONT_DECLARE(roboto18) LV_FONT_DECLARE(bigicons)"
	; -D CONFIG_LV_FONT_DEFAULT="&roboto14"
  -D TEST_AMP=&a
lib_deps =
  ; Use direct URL, because package registry is unstable
  ;lvgl@~7.11.0
  lvgl=https://github.com/lvgl/lvgl/archive/refs/tags/v8.2.0.zip
lib_archive = false

[env:emulator_64bits]
platform = native@^1.1.3
extra_scripts = support/sdl2_build_extra.py
build_flags =
  ${env.build_flags}
  ; -D LV_LOG_LEVEL=LV_LOG_LEVEL_INFO
  -D LV_LOG_PRINTF=1
  ; Add recursive dirs for hal headers search
  !python -c "import os; print(' '.join(['-I {}'.format(i[0].replace('\x5C','/')) for i in os.walk('hal/sdl2')]))"
  -lSDL2
  ; SDL drivers options
  -D LV_LVGL_H_INCLUDE_SIMPLE
  -D LV_DRV_NO_CONF
  -D USE_SDL
  -D SDL_HOR_RES=480
  -D SDL_VER_RES=320  
  -D SDL_ZOOM=1
  -D SDL_INCLUDE_PATH="\"SDL2/SDL.h\""

  ; LVGL memory options, setup for the demo to run properly
  -D LV_MEM_CUSTOM=1
  -D LV_MEM_SIZE="(128U * 1024U)"
  
lib_deps =
  ${env.lib_deps}
  ; Use direct URL, because package registry is unstable
  ;lv_drivers@~7.9.0
  lv_drivers=https://github.com/lvgl/lv_drivers/archive/refs/tags/v8.2.0.zip
src_filter =
  +<*>
  +<../hal/sdl2>
  ; Force compile LVGL demo, remove when working on your own project
  +<../.pio/libdeps/emulator_64bits/lvgl/demos>

[env:emulator_32bits]
extends = env:emulator_64bits
build_flags =
  ${env:emulator_64bits.build_flags}
  -m32
src_filter =
  +<*>
  +<../hal/sdl2>
  +<../.pio/libdeps/emulator_32bits/lvgl/demos>

[env:stm32f429_disco]
platform = ststm32@^8.0.0
board = disco_f429zi
framework = stm32cube
build_flags =
  ${env.build_flags}
  -D LV_LOG_LEVEL=LV_LOG_LEVEL_NONE
  ; header's default is 25MHz, but board uses 8MHz crystal
  -D HSE_VALUE=8000000
  ; Add recursive dirs for hal headers search
  !python -c "import os; print(' '.join(['-I {}'.format(i[0].replace('\x5C','/')) for i in os.walk('hal/stm32f429_disco')]))"
lib_deps =
  ${env.lib_deps}
  BSP-ili9341
  BSP-stmpe811
src_filter =
  +<*>
  +<../hal/stm32f429_disco>
  ; Force compile LVGL demo, remove when working on your own project
  +<../.pio/libdeps/stm32f429_disco/lvgl/demos>
```

### Additional info
I'm not really familiar with pio for now, I don't think the bug is related directly to lvgl, but if so I'm sorry.
I hope I can be of any help

ivankravets, 2022-08-24:
Could you escape as described in https://docs.platformio.org/en/latest/projectconf/section_env_build.html#stringification?

```
'-D TEST_AMP=\\&a'
```

Does it work now?

tartinesKiller, 2022-08-25:
Thanks for your answer!
I tried `'-D TEST_AMP=\\&a'`, but compiler was complaining of a stray `\` in code.
However, I did try 
```
'-D TEST_AMP=&a'
```
and it's working on both Windows and Ubuntu. The space after the `-D` and the simple quotes around the line seems to have their importance. I guess I should have paid attention to this line in the doc:

> Please note that we enclosed the flag in the single quotes to prevent the shell from removing double quotes

But I can't find anything refering to the space after the `-D`. 
Should I close the issue anyway?
Thanks a lot for your help!

ivankravets, 2022-09-27:
Thanks for the issue. Please review updated docs https://docs.platformio.org/en/latest/projectconf/section_env_build.html#stringification

commit ca48e6c1 (2022-09-27): Document a stringification of ampersand // Resolve #4396
closed 2022-09-27 by ivankravets