platformio/platformio-core #4398: Allow to change Include paths for cppcheck

dzid26, 2022-08-20:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 6.1.4
### Description of problem
Lack of ability to exclude Includes from cppcheck analysis. Long analysis and a long output.
Related [conversation](https://community.platformio.org/t/cppcheck-how-to-exclude-includes/29265/5?u=dzidmail).

#### Steps to Reproduce

1. Create project which includes hal headers from within the repository. 
2. Setup cppcheck with misra check: 
3. Perform static analysis  `platformio.exe check`

### Actual Results
Analysis that is slow and outputs thousands of MISRA violations
```
Component     HIGH    MEDIUM    LOW
-----------  ------  --------  -----
src            0        0        4
src\CMSIS      0        0      1173
src\Device     0        0      8279

Total          0        0      9456

Environment         Tool      Status    Duration
------------------  --------  --------  ------------
genericSTM32F103C8  cppcheck  PASSED    00:00:19.038
```

### Expected Results
A quick analysis of user code only without entering hal headers
```
Component     HIGH    MEDIUM    LOW
-----------  ------  --------  -----
src            0        0        4

Total          0        0      4

Environment         Tool      Status    Duration
------------------  --------  --------  ------------
genericSTM32F103C8  cppcheck  PASSED    00:00:1.234
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
board_build.ldscript = src/Device/STM32F103XB_FLASH.ld
build_flags =
    -Isrc/CMSIS/
    -Isrc/Device

check_skip_packages = yes ;don't inlcude compiler package inludes
check_patterns = 
  src/*.c
  src/*.h
check_flags =
  cppcheck:--addon=misra --suppress=suppress.txt
```

**Source file to reproduce issue:**
Working example:
https://github.com/dzid26/pio-exclude-cppcheck.

dzid26, 2022-08-31:
Understood. 
Also, I imagine that for projects using Platformio's frameworks, checking framework headers could be avoided using `check_skip_packages = yes`?

Regarding the headers expansion though, cppcheck [manual](https://cppcheck.sourceforge.io/manual.pdf) discourages including system headers: [image]

valeros, 2022-08-31:
> Also, I imagine that for projects using Platformio's frameworks, checking framework headers could be avoided using check_skip_packages = yes?

That's right. This feature was added as a fall-back option mainly because Cppcheck's preprocessor is simply not capable of parsing some of currently available embedded frameworks. 

> Regarding the headers expansion though, cppcheck [manual](https://cppcheck.sourceforge.io/manual.pdf) discourages including system headers:

PlatformIO supports several static analysis tools and even though Cppcheck may discourage specifying all headers, other tools may rely on complete translation units. Besides, in PlatformIO scope, only the platform packages (toolchains, frameworks, etc.) are considered system headers. That's where `check_skip_packages` can be used to avoid these headers. IMO, once you put a low-level framework inside your project, the headers from that frameworks shouldn't be considered as system ones.