platformio/platformio-core #4400: Failed to update package if using a platform_package pointing to a git's repository

X-Ryl669, 2022-08-23:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: MacOSX

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.4

### Description of problem
I've this in my platformio.ini:
```
...
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = espidf

[env:rvcu]
extends = esp-wrover-kit
platform=${env:esp-wrover-kit.platform}
platform_packages = toolchain-xtensa-esp32@8.4.0+2021r2
                    toolchain-riscv32-esp@8.4.0+2021r2
                    toolchain-xtensa-esp32s2@8.4.0+2021r2
                    framework-espidf@https://github.com/X-Ryl669/esp-idf#allowSPIFromISR

board = esp-wrover-kit
framework = espidf
...
```
If I run `pio run` it downloads the whole git repository and submodules, yet, in `~//.platformio/packages/framework-espidf` is not a git repository (it misses `.git` subfolder). Nor is `framework-espidf@src-04c5efb3c04139a689accdcd494a8a7f` (they are the only 2 folders I have in `~/.platformio/packages` for framework-espidf)

If I run `pio pkg update` I get this error then:
```
pio pkg update
Resolving esp-wrover-kit dependencies...
Already up-to-date.
Resolving rvcu dependencies...
Error: VCS: Could not receive an output from `['git', 'branch']` command ({'out': '', 'err': 'fatal: not a git repository (or any of the parent directories): .git\n', 'returncode': 128})
```

If I'm not using a git repository (not specifying `@https://gitrepo`), it works.  

I wonder if the package install doesn't copy the .git folder correctly?

X-Ryl669, 2022-08-23:
Well, I'm not sure it works when checking out the repository, let me explain:
It states when running pio run on a missing `framework-espidf` folder (after `rm -rf ~/.platformio/packages/framework-espidf*`)
```
$ pio run -v
[...]
Tool Manager: Installing git+https://github.com/X-Ryl669/esp-idf#allowSPIFromISR
git version 2.17.2 (Apple Git-113)
Cloning into '/Users/cyril/.platformio/.cache/tmp/pkg-installing-mk28cgar'...
remote: Enumerating objects: 11641, done.
remote: Counting objects: 100% (11641/11641), done.
remote: Compressing objects: 100% (9200/9200), done.
remote: Total 11641 (delta 2838), reused 5921 (delta 1912), pack-reused 0
Receiving objects: 100% (11641/11641), 64.45 MiB | 900.00 KiB/s, done.
Resolving deltas: 100% (2838/2838), done.
Checking out files: 100% (10052/10052), done.
[...]
Resolving deltas: 100% (3773/3773), done.
Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
Submodule path 'components/bt/controller/lib_esp32': checked out 'c98341759b2c152c58878db6daf22d153a75fda4'
Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out '34e6a82bab1a0d09eb8d96277d8c7b4d9006071e'
Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '3b0038690a644498d6d80f1de8df0efff8cd8cf5'
Submodule path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib': checked out '8bf4c4c2cb1f964122977c9995863f325346411d'
Submodule path 'components/bt/host/nimble/nimble': checked out 'c1cb851c8562d0f1a4d105614c336341e2a535b2'
Submodule path 'components/cmock/CMock': checked out 'eeecc49ce8af123cf8ad40efdb9673e37b56230f'
Submodule 'vendor/c_exception' (https://github.com/throwtheswitch/cexception.git) registered for path 'components/cmock/CMock/vendor/c_exception'
Submodule 'vendor/unity' (https://github.com/throwtheswitch/unity.git) registered for path 'components/cmock/CMock/vendor/unity'
Cloning into '/Users/cyril/.platformio/.cache/tmp/pkg-installing-mk28cgar/components/cmock/CMock/vendor/c_exception'...
remote: Enumerating objects: 335, done.        
remote: Counting objects: 100% (9/9), done.        
remote: Compressing objects: 100% (9/9), done.        
remote: Total 335 (delta 1), reused 1 (delta 0), pack-reused 326        
Receiving objects: 100% (335/335), 555.36 KiB | 920.00 KiB/s, done.
Resolving deltas: 100% (154/154), done.
Cloning into '/Users/cyril/.platformio/.cache/tmp/pkg-installing-mk28cgar/components/cmock/CMock/vendor/unity'...
remote: Enumerating objects: 6110, done.        
remote: Counting objects: 100% (333/333), done.        
remote: Compressing objects: 100% (182/182), done.        
remote: Total 6110 (delta 188), reused 262 (delta 150), pack-reused 5777        
Receiving objects: 100% (6110/6110), 6.70 MiB | 947.00 KiB/s, done.
Resolving deltas: 100% (3772/3772), done.
Submodule path 'components/cmock/CMock/vendor/c_exception': checked out '71b47be7c950f1bf5f7e5303779fa99a16224bb6'
Submodule path 'components/cmock/CMock/vendor/unity': checked out 'cf949f45ca6d172a177b00da21310607b97bc7a7'
Submodule path 'components/esp_phy/lib': checked out '8d3ce02436066e33e8964d34981a97fe64edae2e'
Submodule path 'components/esp_wifi/lib': checked out '845af059ecbb64cb175df20cd6e61d956fdf2a66'
Submodule path 'components/heap/tlsf': checked out 'ff11688f242b28b3918c2cdaa20738d12d73b5f4'
Submodule path 'components/ieee802154/lib': checked out 'd9838acadd72e5021cee5f79a897e195076d3645'
Submodule path 'components/json/cJSON': checked out 'd348621ca93571343a56862df7de4ff3bc9b5667'
Submodule path 'components/lwip/lwip': checked out 'ec115c720b8909364ec5898dc5cc430cc898192d'
Submodule path 'components/mbedtls/mbedtls': checked out '9bb5effc3298265f829878825d9bd38478e67514'
Submodule path 'components/mqtt/esp-mqtt': checked out 'ae53d799da294f03ef65c33e88fa33648e638134'
Submodule path 'components/openthread/lib': checked out 'af1ac2b3770335db96702bcb06ca793a1425dc86'
Submodule path 'components/openthread/openthread': checked out 'c055c10a9a1362a96687614d4478dd2773e6ddc0'
Submodule path 'components/protobuf-c/protobuf-c': checked out 'f224ab2eeb648a818eb20687d7150a285442c907'
Submodule path 'components/spiffs/spiffs': checked out '0dbb3f71c5f6fae3747a9d935372773762baf852'
Submodule path 'components/tinyusb/tinyusb': checked out 'c1986586098643edfa6b75a776299c576ad0ce8b'
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Tool Manager: framework-espidf@1.1.0+sha.cf1fc16 has been installed!
```

But if I list the content of the `~/.platformio/packages/framework-espidf`, it doesn't contain the .git repository, and even worse, it doesn't contain the repository at all, but some unrelated files????
```
$ cd ~/.platformio/packages/framework-espidf/
$ ls
README.md		cjson-utils-tests.json	components		package.json		spec_tests.json		tests.json		version.txt
$ cat README.md => this is cJSON repository and not espidf!!!
```

ivankravets, 2022-08-24:
> https://github.com/X-Ryl669/esp-idf#allowSPIFromISR

This is not a compatible package. See https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#package-specifications

closed 2022-08-24 by ivankravets

X-Ryl669, 2022-08-24:
So the silent failure is expected ? 
Shouldn't the process report that the given URL doesn't contain a `package.json` so at least it gives a clue to what's wrong ?

ivankravets, 2022-08-25:
PlatformIO Core scans for manifest files recursively following nested folders. It seems there is `pakcage.json` in some folders. A solution to this issue is to create a `package.json` in the root of repository.