platformio/platformio-core #4403: `-U name` in build_flags not works as as describute in doc

yyjdelete, 2022-08-29:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: win10_x64

**PlatformIO Version** (`platformio --version`): core 6.1.4 + vscode 1.70.2 + PlatformIO IDE for VSCode v2.5.2

### Description of problem
`-U` in build_flags not works as as describute in doc
https://docs.platformio.org/en/latest/projectconf/section_env_build.html#scopes-scons-variables
The doc say there is an space between `-U` and flags, but unlike what works with `-D`,  `-U` doesn't really work with the space added, but work well without the space. 

Trace show platformio just ***pass `-U` but without actual flags to gcc***, which lead to the error
```
C:\WINDOWS\System32\cmd.exe /C "arm-none-eabi-gcc -o ".pio\build\tag\src\xxxx.o" -c -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -O0 -ggdb3 -g3 -U -DF_CPU=64000000L -DPLATFORMIO=60104 -DSTM32G0xx -DSTM32G070xx -DNDEBUG -DTX_NODE=1 -D__PLATFORMIO_BUILD_DEBUG__ -DDEBUG -U -ICore\Inc -I. "xxxx.c""
```
***Try to remove the space between `-U` and flags works well***.(-U {flag} => -U{flag})
There must be something wrong with the doc or the actual implementation.

#### Steps to Reproduce

1. build

### Actual Results
```
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
<command-line>:0:1: error: macro names must be identifiers
<command-line>:0:2: warning: extra tokens at end of #undef directive
```

### Expected Results
build success, and it work like `-O0 -ggdb3 -g3 -D DEBUG`(NDEBUG is undef in debug builds)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
build_flags = 
    -D NDEBUG
    -U NDEBUG2
debug_build_flags = -O0 -ggdb3 -g3 -D DEBUG -U NDEBUG
build_type = debug
```

**Source file to reproduce issue:**
```cpp
```

### Additional info

mcspr, 2022-08-31:
[Quoting seems to help](https://github.com/platformio/platformio-core/blob/c56b35f5047d1a8c6837e30e6333ad22f595b783/platformio/builder/tools/piobuild.py#L117)
```ini
build_flags = 
    "-D NDEBUG"
    "-U NDEBUG2"
```

Does not look like this comes from PIO, but from SCons itself and the way it treats cmdline. And don't see anything in the [Environment.py](https://github.com/SCons/scons/blob/e3e91c16087dc50cfb5205eb0142e0877a5383bd/SCons/Environment.py#L785) source dealing with -U.

What happens instead is `ParseFlags()` removes `TEST` (which would have been added to CCLFAGS as-is, right after `-U`), and for some reason treats it as a file input and adds it to the `LIBS` array. 
```python
def show(string):
    env = Environment()
    flags = env.ParseFlags(string)

    print(string)
    for key, value in flags.items():
        if not value:
            continue
        print(f"{key=} {value=}")

show("-DTEST -UTEST")
show("-D TEST -U TEST")
show('"-D TEST" "-U TEST"')
```
```bash
> pio pkg exec -p tool-scons scons.py
Using tool-scons@4.40400.0 package
-DTEST -UTEST
key='CCFLAGS' value=['-UTEST']
key='CPPDEFINES' value=['TEST']
-D TEST -U TEST
key='CCFLAGS' value=['-U']
key='CPPDEFINES' value=['TEST']
key='LIBS' value=[<SCons.Node.FS.File object at 0x5637b31528f0>]
"-D TEST" "-U TEST"
key='CCFLAGS' value=['-U TEST']
key='CPPDEFINES' value=[' TEST']
scons: `.' is up to date.
```

Also comparing to GCC
```console
> echo | gcc -D TEST -E -dM - | grep 'define TEST'
#define TEST 1
> echo | gcc -D TEST -U TEST -E -dM - | grep 'define TEST'
(no define, flags cancel each other out)
```

Which I guess makes this a SCons bug?

ivankravets, 2023-04-21:
Yes, a similar issue to https://github.com/platformio/platformio-core/issues/4541

I explained in the comment regarding SCons and POSIX/Shell argument parsing. Using quotes should help.

closed 2023-04-21 by ivankravets