platformio/platformio-core #4407: UnboundLocalError: local variable 'port' referenced before assignment in finder.py

mrfaptastic, 2022-09-01:
What kind of issue is this? 

PlatformIO Core

------------------------------------------------------------------

### Configuration

Windows 10 64bit

**PlatformIO Version**

6.1.4

### Description of problem

I tried to write firmware to an ESP32 device, but had not installed the cp2102 driver on my windows 10 device - which exposed a hidden bug in \.platformio\penv\lib\site-packages\platformio\device\finder.py

It's checking for the value of 'port' but in this edge-case, 'port' variable is never initialised. So finder.py errors out with a programmer error opposed to something saying "can't find a port".

#### Steps to Reproduce

1. Don't have any Serial to USB driver.
2. Attempt to burn firmware to device using PIO
3. <del>Profit</del> finder.py errors with a python coding error.

### Actual Results

```
Building .pio\build\esp32-s2-saola-1\firmware.bin
esptool.py v3.3
Creating esp32s2 image...
Merged 2 ELF sections
Successfully created esp32s2 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
*** [upload] UnboundLocalError : local variable 'port' referenced before assignment
Traceback (most recent call last):
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Action.py", line 1318, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\Anonymous\.platformio\platforms\espressif32\builder\main.py", line 35, in BeforeUpload
    env.AutodetectUploadPort()
  File "C:\Users\Anonymous\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Util.py", line 737, in __call__
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anonymous\.platformio\penv\lib\site-packages\platformio\builder\tools\pioupload.py", line 113, in AutodetectUploadPort
    UPLOAD_PORT=SerialPortFinder(
  File "C:\Users\Anonymous\.platformio\penv\lib\site-packages\platformio\device\finder.py", line 156, in find
    return best_port or port
UnboundLocalError: local variable 'port' referenced before assignment
========================================================================== [FAILED] Took 43.65 seconds ==========================================================================

 *  The terminal process "C:\Users\Anonymous\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

### Expected Results

An error message about how a serial port can't be found.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

Not relevant, but here it is anyway:

```ini
[env:esp32-s2-saola-1]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
monitor_speed = 115200
upload_speed = 921600

```

**Source file to reproduce issue:**

In  .platformio\penv\lib\site-packages\platformio\device\finder.py

```python
    def find(self, initial_port=None):
          if initial_port:
            if not is_pattern_port(initial_port):
                return initial_port
            return self.match_serial_port(initial_port)

        if self.upload_protocol and self.upload_protocol.startswith("blackmagic"):
            return self._find_blackmagic_port()

        device = None
        if self.board_config and self.board_config.get("build.hwids", []):
            device = self._find_board_device()
        if not device:
            device = self._find_known_device()
        if device:
            port = self._reveal_device_port(device)

        # pick the best PID:VID USB device
        best_port = None
        for item in list_serial_ports():
            if self.ensure_ready and not is_serial_port_ready(item["port"]):
                continue
            port = item["port"]
            if "VID:PID" in item["hwid"]:
                best_port = port
        return best_port or port

```

### Additional info

Declare the port variable in the 'find' function first...

```python
    def find(self, initial_port=None):
        port = None <------- ADD THIS HERE
          if initial_port:
            if not is_pattern_port(initial_port):
                return initial_port
            return self.match_serial_port(initial_port)

        if self.upload_protocol and self.upload_protocol.startswith("blackmagic"):
            return self._find_blackmagic_port()

        device = None
        if self.board_config and self.board_config.get("build.hwids", []):
            device = self._find_board_device()
        if not device:
            device = self._find_known_device()
        if device:
            port = self._reveal_device_port(device)

        # pick the best PID:VID USB device
        best_port = None
        for item in list_serial_ports():
            if self.ensure_ready and not is_serial_port_ready(item["port"]):
                continue
            port = item["port"]
            if "VID:PID" in item["hwid"]:
                best_port = port
        return best_port or port

```

ivankravets, 2022-09-03:
Please re-test with `pio upgrade --dev`.

closed 2022-09-03 by ivankravets

JonAdamsFromNC, 2022-09-21:
I had the same issue, just now, with a M5Stack Core board. Interestingly it wrote to the board fine *before* installing their "M5Unified" Library, but failed after installing that library?
Upgrading the PIO installation to dev as noted addressed the issue for me.

szendro, 2022-09-25:
I had the same problem and took some head banging to resolve:
Platform IO should find the port automatically so no need edit ini file.
The problem turned out to be that a driver is required to convert USB to serial.
Go to Silicon Labs website and download the generic driver for CP210x and follow the installation instructions.
Platformio should now be able to find the port.
Good luck!

italocjs, 2022-10-03:
I had this error with a bad usb cable, the error was "UnboundLocalError: local variable 'port' referenced before assignment" i guess that if this is the default error for no port found, mayber a more clearer message would be helpful.

commit e0a3b818 (2022-09-03): Fixed "UnboundLocalError: local variable 'port' referenced before assignment" // Resolve #4407