platformio/platformio-core #4410: Cannot upgrade platformio lib to pio pkg to install library EspSoftwareSerial

Erriez, 2022-09-04:
### Configuration

**Operating system**: Ubuntu 22.04

**PlatformIO Version**: PlatformIO Core, version 6.1.4

### Description of problem

I'm porting CI build scripts from legacy `platformio lib` to `pio pkg install`, but does not work.

Background: The developer of library `EspSoftwareSerial@6.16.1` introduced a naming conflict in `src/SoftwareSerial.[c|h]` as this library can only be used for ESP targets, not AVR. For this reason, it cannot be installed globally. A workaround was to install this library directly in the ESP libraries framework directory: `.platformio/packages/framework-arduinoespressif32/libraries/EspSoftwareSerial/`.

**Legacy code**:
```
$ platformio lib --storage-dir ~/.platformio/packages/framework-arduinoespressif32/libraries install "EspSoftwareSerial@6.16.1"

Install ESPSoftwareSerial into framework-arduinoespressif32 to prevent conflicts with generic name SoftwareSerial

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg install` instead.

Library Manager: Installing EspSoftwareSerial @ 6.16.1
Unpacking  [####################################]  100%
Library Manager: EspSoftwareSerial@6.16.1 has been installed!
```

**New code**:

```
$ pio pkg install --storage-dir ~/.platformio/packages/framework-arduinoespressif32/libraries --library "EspSoftwareSerial@6.16.1"

Error: Not a PlatformIO project. `platformio.ini` file has not been found in current working directory
(/home/user/Arduino/libraries/ErriezLibrary). 
To initialize new project please use `platformio project init` command
```

#### Steps to Reproduce

Execute the commands above.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
Not applicable.

**Source file to reproduce issue:**
See above.

Any suggestions how to proceed?

maxgerhardt, 2022-09-07:
>$ pio pkg install --storage-dir ~/.platformio/packages/framework-arduinoespressif32/libraries --library "EspSoftwareSerial@6.16.1"

As `pio pkg install -h` says
```
Usage: pio pkg install [OPTIONS]

Options:
  -d, --project-dir DIRECTORY
  -e, --environment TEXT
  -p, --platform SPECIFICATION
  -t, --tool SPECIFICATION
  -l, --library SPECIFICATION
  --no-save                     Prevent saving specified packages to
                                `platformio.ini`
  --skip-dependencies           Skip package dependencies
  -g, --global                  Install package globally
  --storage-dir DIRECTORY       Custom Package Manager storage for global
                                packages
  -f, --force                   Reinstall package if it exists
  -s, --silent                  Suppress progress reporting
  -h, --help                    Show this message and exit.
  ```
so you should be using

```
pio pkg install --global --storage-dir ~/.platformio/packages/framework-arduinoespressif32/libraries --library "EspSoftwareSerial@6.16.1"
```
as the command instead.

Erriez, 2022-09-07:
Thanks @maxgerhardt for pointing to this package installation command. This solved my problem.

A combination of options `--storage-dir` **and** `--global` is required and I was not aware of this.

maxgerhardt, 2022-09-07:
I guess if the `pio pkg install` command wants to remain backwards compaibilitiy, it should auto-enabled `--global` when `--storage-dir` is given.

ivankravets, 2022-09-07:
Yes, @maxgerhardt, I marked this issue as a bug. We will fix it.

ivankravets, 2022-09-30:
Please re-rest with the latest PlatformIO Core using `pio upgrade --dev`. Thanks.

closed 2022-09-30 by ivankravets

Erriez, 2022-10-01:
@ivankravets The `pio upgrade --dev` works with the command `pio pkg install --storage-dir ~/.platformio/packages/framework-arduinoespressif32/libraries --library "EspSoftwareSerial@6.16.1"`.

Thanks for your support. :+1:

commit e0f9cb8c (2022-09-30): Fixed an issue when `pio pkg install --storage-dir` command requires PlatformIO project // Resolve #4410