platformio/platformio-core #4411: Crash using call to delayMicroseconds

FraserElectronics, 2022-09-07:
### Configuration

**Operating system**:
ESP32 (LILYGO® T-Micro32 Plus 8MB flash 2MB PSRAM)

**PlatformIO Version** (`platformio --version`):
6.1.4

### Description of problem
I am creating a pulse train using the following code:

```
    for (int i = 0; i < steps; i++)
    {
      digitalWrite(STEP_OUTPUT, false);
      delayMicroseconds(speedDelayus);      
      digitalWrite(STEP_OUTPUT, true);
      delayMicroseconds(speedDelayus);
    }
```
speedDelayus is an integer.

This always crashes with:

```
ELF file SHA256: 0000000000000000

Backtrace: 0x400885d8:0x3ffbf8f0 0x40088855:0x3ffbf910 0x4012fe84:0x3ffbf930 0x40086efd:0x3ffbf950 0x40084242:0x3ffccd30 0x40084e7b:0x3ffccd50 0x4008119f:0x3ffccd70 0x400811df:0x3ffccd90 0x400d1585:0x3ffccdb0 0x400d0ab3:0x3ffccdd0 0x400d0d19:0x3ffccdf0 0x400d12d5:0x3ffcce50 0x40089866:0x3ffcce70

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
```

If I replace delayMicroseconds with delay, it works every time.

#### Steps to Reproduce

1. Try calling delayMicroseconds in a loop.

### Actual Results
See above.

### Expected Results
For it not to crash.

Any help would be much appreciated as I need microsecond resolution for my pulse timings.

Andy

FraserElectronics, 2022-09-07:
Further to this, I am actually using delayMicroseconds in a task and I not that delay calls vTaskDelay whereas delayMicroseconds  does not - is this my problem?

Mkoopman828, 2022-09-08:
From your elf output it seems that it is not the abstract calling at fault.