platformio/platformio-core #4414: LDF doesn't evaluate macro properly

maxgerhardt, 2022-09-12:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`): 6.1.5a3

### Description of problem

If the Arduino-ESP32's SD_MMC library is included and LDF mode deep+ or chain+ are used, the subdependency to the FS library is not detected and the build fails. PlatformIO's LDF fails to analyze the `#ifdef` macro correctly.

#### Steps to Reproduce

1. Create a new `featheresp32` project
2. Add `lib_ldf_mode = chain+` to `platformio.ini`
3. Add `#include <SD_MMC.h>` to `src/main.cpp`
4. Press Build Button

### Actual Results

```
Dependency Graph
|-- SD_MMC @ 2.0.0
Building in release mode
[..]
*** [.pio\build\featheresp32\libd79\SD_MMC\SD_MMC.cpp.o] Error 1
In file included from src/main.cpp:2:
C:/Users/Max/.platformio/packages/framework-arduinoespressif32/libraries/SD_MMC/src/SD_MMC.h:21:10: fatal error: FS.h: No such file or directory

************************************************************
* Looking for FS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:FS.h"
* Web  > https://registry.platformio.org/search?q=header:FS.h
*
************************************************************

 #include "FS.h"
          ^~~~~~
compilation terminated.
*** [.pio\build\featheresp32\src\main.cpp.o] Error 1
```

### Expected Results

Successfull compilation.

```
Dependency Graph
|-- SD_MMC @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
[...]

```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:featheresp32]
platform = espressif32
board = featheresp32
lib_ldf_mode = chain+
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <SD_MMC.h>

void setup() {}
void loop() {}
```

### Additional info

The build always passes with the "dumb" LDF modes that don't try to evaluate macros (`chain` and `deep`). 

[The library](https://github.com/espressif/arduino-esp32/blob/37cbaeccb7684c2a3c787819adeccf659a5951f2/libraries/SD_MMC/src/SD_MMC.h#L17-L21) conditionally includes `FS.h` based on 
```cpp
#include "soc/soc_caps.h"
#ifdef SOC_SDMMC_HOST_SUPPORTED            
#include "FS.h"
```

The `deep+` and `chain+` mode should correctly detect that the macro `SOC_SDMMC_HOST_SUPPORTED` is enabled (in `soc/soc_caps.h`) and detect the `FS.h` inclusion, but it does not.

This is showcased by adding `build_flags = -DSOC_SDMMC_HOST_SUPPORTED` to the `platformio.ini`, which makes the build pass with `deep+` and `chain+` modes.

HamzaHajeir, 2023-05-11:
I'm facing a similar issue, I can't build even with deep+ ldf_mode...

StKob, 2025-03-26:
Looks like its still not fixed.

salikhafeez44, 2025-06-15:
A workaround is adding

build_flags = -DSOC_SDMMC_HOST_SUPPORTED
to the platformio.ini, then deep+ mode can be used.