platformio/platformio-core #4416: FEAT: Configurable delay after Upload but before Monitor

ByronAP, 2022-09-16:
When using a Teensy board Upload and Monitor fails.

I do not feel like this is a bug just a limitation due to how the Teensy is written to.

Since programming a Teensy is done outside of PIO there is no exact way to tell if the upload has actually completed and the board reset.
I think adding a configuration option that specifies an amount of time to wait between upload and monitor would help this and other issues I have encountered.

ALA:
`monitor_delay: 10000`

Which would cause a 10 second delay before starting the serial monitor.

maxgerhardt, 2022-09-16:
Effectively duplicate of https://github.com/platformio/platformio-core/issues/3742, but only a workaround is given with some extra script, no official integration into the core as a feature.

ByronAP, 2022-09-16:
Not as elegant as bultin delay functions but since it is a solution I will close this, thank you

closed 2022-09-16 by ByronAP

maxgerhardt, 2022-09-16:
I would actually like to see this implemented in the core for more robustness in these USB-serial cases, maybe @ivankravets has an opinion on that.

ByronAP, 2022-09-16:
for anyone coming across this, here is the extra script I used

```
import time

Import("env")

def after_upload(source, target, env):
    print("Sleeping for 5 seconds.........................................................................................")
    print("Start : %s" % time.ctime())
    time.sleep(5)
    print("End : %s" % time.ctime())

# setting the build to debug makes issues with teensy so to check if we are in debug check for the word 'debug' in the environment name
# you should change this or remove it if you do not use the same naming conventions (teensy41-debug, teensy41-release)
if "debug" in env["PIOENV"]:
    env.AddPostAction("upload", after_upload)
```

ByronAP, 2022-09-17:
To expand on this. I wanted to have this set from the pio ini file so I added a field to the environment
`monitor_delay: 6`
making it easy to change the delay time from within the config. 

The python script now is this
```
import time

Import("env")

def after_upload(source, target, env):
    delaySeconds = int(env.GetProjectOption("monitor_delay"))
    print("Sleeping for ", delaySeconds, " seconds")
    for x in range(delaySeconds):
        time.sleep(1)
        print(delaySeconds - x)

    print("READY")

# setting the build to debug makes issues with teensy so to check if we are in debug check for the word 'debug' in the environment name
# you should change this or remove it if you do not use the same naming conventions (teensy41-debug, teensy41-release)
if "debug" in env["PIOENV"]:
    env.AddPostAction("upload", after_upload)
```

maxgerhardt, 2022-09-17:
Just for my interest, the script in https://github.com/platformio/platformio-core/issues/3742#issuecomment-1003454439 that waits for the previously found monitor port to re-appear, does it work better? No fine-tuned delay options then.

ByronAP, 2022-09-17:
I felt like it went beyond the scope of the problem (inability to know if teensy upload is complete) and that it created the potential for more problems than it solves along with added complexity and offers a smaller use case. Just my 2c though

maxgerhardt, 2022-09-17:
But it also hooks the "after upload" function, so upload is compeleted at that point.

ByronAP, 2022-09-17:
yes, but I did not find a way to detect if monitor was selected, so if I disconnect the board, upload fails, reboot fails, board gets stuck and a few other cases it will wait potentially indefinitely.

ByronAP, 2022-09-17:
I think this is because after upload in the case of teensy is really after file copy and teensy doesn't return a true success. The com thing is nice in its own right but would need a time limit to let it bailout.

ByronAP, 2022-09-17:
ok so I changed things up a bit to offer best of both worlds

these are the config options used
```
monitor_speed = 115200
monitor_port = COM4
monitor_port_timeout = 6
monitor_delay = 6
extra_scripts = post:extra_script.py
```
and this is the script
```
import datetime as dt
import serial

Import("env")

def do_sleep(sleepSeconds):
    for x in range(sleepSeconds):
        dt.sleep(1)
        print(sleepSeconds - x)

def after_upload(source, target, env):
    serialPort = None
    portTimeout = 15
    delaySeconds = 0
    portSpeed = 9600

    try:
        serialPort = env.GetProjectOption("monitor_port")
    except:
        pass

    try:
        portTimeout = int(env.GetProjectOption("monitor_port_timeout"))
    except:
        pass

    try:
        delaySeconds = int(env.GetProjectOption("monitor_delay"))
    except:
        pass

    try:
        portSpeed = int(env.GetProjectOption("monitor_speed"))
    except:
        pass

    if serialPort is not None:
        print("Waiting on " + serialPort + "...")

        endTime = dt.datetime.now(dt.timezone.utc) + \
            dt.timedelta(seconds=(portTimeout*2))

        while True:
            try:
                s = serial.Serial(serialPort, portSpeed,
                                  timeout=(portTimeout/2))
                break
            except:
                pass

            if dt.datetime.now(dt.timezone.utc) > endTime:
                print("ERR: Timeout waiting on " + serialPort)
                break
    else:
        if delaySeconds > 0:
            print("Sleeping for " + delaySeconds + " seconds")
            do_sleep(delaySeconds)
        else:
            print("No monitor_delay or monitor_port configured")

    print("READY")

# setting the build to debug makes issues with teensy so to check if we are in debug by checking for the word 'debug' in the environment name
# you should change this or remove it if you do not use the same naming conventions (teensy41-debug, teensy41-release)
if "debug" in env["PIOENV"]:
    env.AddPostAction("upload", after_upload)
```