platformio/platformio-core #4423: The system cannot find the file specified

davidjmstewart, 2022-09-24:
I am getting this issue (replicated in full below)when trying to run tests for my project (unity has been successfully installed). Related to having a space in the path name?

I am on PlatformIO Core, version 6.1.4

**Full error output**
```
 Executing task: C:\Users\David Stewart\.platformio\penv\Scripts\platformio.exe test --verbose --environment QUTy 

Collected 1 tests (*)

Processing * in QUTy environment
------------------------------------------------------------------------------------------------------------------
Building...

Testing...
[WinError 2] The system cannot find the file specified
--------------------------------------- QUTy:* [ERRORED] Took 1.63 seconds ---------------------------------------

==================================================== SUMMARY ====================================================
Environment    Test    Status    Duration
-------------  ------  --------  ------------
QUTy           *       ERRORED   00:00:01.627
=================================== 1 test cases: 0 succeeded in 00:00:01.627 ===================================

 *  The terminal process "C:\Users\David Stewart\.platformio\penv\Scripts\platformio.exe 'test', '--verbose', '--environment', 'QUTy'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

**Test file**

```C
#include <stdio.h>
#include <stdbool.h>
#include "unity.h"

void setUp(void)
{
    // set stuff up here
}

void tearDown(void)
{
    // clean stuff up here
}

void test_ex(void)
{
    UNITY_TEST_ASSERT_EQUAL_UINT(1, 2, 17, "Intentionally failing test to sanity check the test setup");    
}

int main(void)
{
    UNITY_BEGIN();
    RUN_TEST(test_ex);
    return UNITY_END();
}
```

**Test folder structure**

The source file with the tests in it is directly under the `test/` directory in the project.
[image]

davidjmstewart, 2022-09-25:
>Related to having a space in the path name?

I don't think this is the case: I tested on another machine without a space in the path name and the same issue presents.

ivankravets, 2022-09-25:
Do you use any espionage software (antivirus tools)? Could you try to disable them or exclude project folder?

davidjmstewart, 2022-09-25:
Just Windows defender on both machines i've tried on. I've tried adding the project directory (still no change) and the project directory + the tests directory (still nothing):

[image]

maxgerhardt, 2022-09-27:
is the project within a OneDrive? I've seen the user's `Documents` folder being redirected by Windows to be the OneDrive folder if it was installed. OneDrive interacts extremely badly with PlatformIO, try moving the project folder to another place.

davidjmstewart, 2022-09-27:
No, not in any form of automatic cloud backup (OneDrive, Google Drive, Dropbox etc).

maxgerhardt, 2022-09-27:
What's the `platformio.ini`? This error can also occur when the maximum Widnows filepath length is exhausted and "Long Win32 paths" is not enabled.

davidjmstewart, 2022-09-27:
I've just tested on another machine that has the project in `C:\dev\path_to_project_here\` and added that path to the Windows antivirus exclusion list - same issue persists.

davidjmstewart, 2022-09-27:
**platformio.ini**

```
[env:QUTy]
platform = quty
board = QUTy
build_flags =
    -L lib/FatFs/
    -I include/FatFs/
    -I lib/FatFs/ff.c
    -Wno-misleading-indentation
    -Werror
    -Wall
```

maxgerhardt, 2022-09-27:
I see an open PR at https://github.com/cab202/quty/pull/1. Does your error message change when you use
```
platform = https://github.com/Tarang74/quty.git
```
?

davidjmstewart, 2022-09-27:
No, the only thing that changed was that an initialisation step was performed. 

**Error message with `platform = quty`**

```
  Executing task: C:\Users\David Stewart\.platformio\penv\Scripts\platformio.exe test --environment QUTy 

Verbosity level can be increased via `-v, -vv, or -vvv` option
Collected 1 tests

Processing * in QUTy environment
--------------------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
[WinError 2] The system cannot find the file specified
------------------------------------------------- QUTy:* [ERRORED] Took 0.88 seconds -------------------------------------------------

============================================================== SUMMARY ==============================================================
Environment    Test    Status    Duration
-------------  ------  --------  ------------
QUTy           *       ERRORED   00:00:00.879
============================================= 1 test cases: 0 succeeded in 00:00:00.879 ============================================= 

 *  The terminal process "C:\Users\David Stewart\.platformio\penv\Scripts\platformio.exe 'test', '--environment', 'QUTy'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

**Error message with platform = https://github.com/Tarang74/quty.git**

```
 *  Executing task: C:\Users\David Stewart\.platformio\penv\Scripts\platformio.exe test --environment QUTy 

Verbosity level can be increased via `-v, -vv, or -vvv` option
Collected 1 tests

Processing * in QUTy environment
--------------------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
[WinError 2] The system cannot find the file specified
------------------------------------------------- QUTy:* [ERRORED] Took 1.79 seconds -------------------------------------------------

============================================================== SUMMARY ============================================================== 
Environment    Test    Status    Duration
-------------  ------  --------  ------------
QUTy           *       ERRORED   00:00:01.786
============================================= 1 test cases: 0 succeeded in 00:00:01.786 ============================================= 

 *  The terminal process "C:\Users\David Stewart\.platformio\penv\Scripts\platformio.exe 'test', '--environment', 'QUTy'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

davidjmstewart, 2022-09-27:
Do you happen to know which file(s) it cannot find? I tried the different verbosity options but I couldn't determine which file in particular was the issue.

maxgerhardt, 2022-09-27:
Aha. On Linux, it directly gives away what it cannot find.

```´Testing...
[Errno 2] No such file or directory: '/home/max/temp/quty/.pio/build/QUTy/program'
```
which seems to be at least referenced in https://github.com/cab202/quty/blob/ea7a16e5dde4452f20d82334d06aa60213ce791c/builder/main.py#L148-L150.

PIO might be confusing itself on what the test firmware to upload is.

maxgerhardt, 2022-09-27:
Giving it that file just make PlatformIO try and execute that binary as if it was a desktop-executable. I think it's mistaking it for a native test and doesn't realize it needs to upload it to a microcontroller.
```
max@PC-LT-23-VM:~/temp/quty$ cp .pio/build/QUTy/firmware.elf .pio/build/QUTy/program
max@PC-LT-23-VM:~/temp/quty$ pio test --without-building -vvv
Collected 1 tests (test_embedded)

Processing test_embedded in QUTy environment
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Testing...
[Errno 8] Exec format error: '/home/max/temp/quty/.pio/build/QUTy/program'
------------------------------------------------------------ QUTy:test_embedded [ERRORED] Took 0.00 seconds ------------------------------------------------------------
```
I'm not yet sure what lines of code in the platform or the PlatformIO core control this behavior.

maxgerhardt, 2022-09-27:
Aha, adding
```py
max@PC-LT-23-VM:~/temp/quty$ cat ~/.platformio/platforms/quty/platform.py
from platformio.public import PlatformBase

class QutyPlatform(PlatformBase):
    def is_embedded(self):
        return True
```
seems to get PlatformIO more on track and ask for the Unity configuration and try to upload it.
```
UnitTestSuiteError: Could not find Unity configuration for the `None` framework.
..
Uploading stage has failed, see errors above. Use `pio test -vvv` option to enable verbose output.
```
I'll have to look at this later, but right now it looks like 1. misconfigured / missing `platform.py` when there should be one and 2. the user project is missing the unity config and code to glue `unity_output_char` etc to the UART output registers -- that code has to come from somewhere for unity to be able to output anything, see [docs](https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/unity.html).

maxgerhardt, 2022-09-27:
@davidjmstewart I've setup https://github.com/maxgerhardt/quty_unittest as a base project that referenced the fixed platform and adds a null-implementation to the needed unity functions. It compiles and PlatformIO wants to upload it. Try and make it work by giving it the actually needed UART implementation and test on your board.

davidjmstewart, 2022-09-27:
Thanks a bunch, I really appreciate it. I'll take a look at this tomorrow :)

maxgerhardt, 2022-10-07:
@davidjmstewart any updates? 

I also think we can move discussion to a new issue in the faulty third-party platform-quty instead of the PlatformIO core.

closed 2022-11-03 by ivankravets