platformio/platformio-core #4432: Correct use of corporate proxy

LioRei, 2022-10-03:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.4

### Description of problem

Platformio is not able to correctly install behing a corporate proxy. HTTPClient and FileDownloader isn't reading env variable to see if there is a proxy or not so the download of platformio/contrib-piohome is not possible.

#### Steps to Reproduce

1. have a computer behing a corporate proxy
2. install platformio
3. try to do a "pio home"

### Actual Results

Error in download of the package

### Expected Results

No error 

### Additional info
I've modified the python script on my computer and I will link a draft PR with my change

LioRei, 2022-10-03:
draft PR: https://github.com/platformio/platformio-core/issues/4432

ivankravets, 2022-10-26:
Could you re-test with `pio upgrade --dev`? See newly added docs on how to configure the proxy https://docs.platformio.org/en/latest/core/installation/proxy-configuration.html

commit 4be5185e (2022-10-26): Added a new ``enable_proxy_strict_ssl`` setting to disable the proxy server certificate verification // Resolve #4432
closed 2022-10-26 by ivankravets