platformio/platformio-core #4434: pio run -t compiledb does not add `-Ilib` to the compile_commands.json

jiriks74, 2022-10-04:
### Configuration

**Operating system**: ArchLinux

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 6.1.4

### Description of problem
Running `platformio run -t compiledb`  to produce `compile_commands.json` does not produce `-Iinclude` in the said file.

This creates "ghost" errors in editors using this file to detect include paths.

#### Steps to Reproduce

1. Create new project
2. Run `pio run -t compiledb`
3. Create a header file in `lib/somefolder/` directory
4. Open editor and include the header from `somefolder` directory
5. You'll get `file not found` error

### Actual Results
No `-Ilib` in the `compile_commands.json`

### Expected Results
`-Ilib` in the `compile_commands.json`

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:uno]
platform = atmelavr
board = uno
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

#include "test.h"

void setup(){
Serial.begin(9600);
}

void loop(){
}
```

### Additional info
[image]

maxgerhardt, 2022-10-05:
>pio run -t compiledb does not add -Ilib 

Correct, PIO never does that for a regular library, it adds `lib/<name of library>` to the include path when it finds a dependency to that library, not the entire `lib/` folder.

When I create
```
\compiledb_test>tree /f
Folder PATH listing for volume Windows
Volume serial number is 10FE-FCFE
C:.
│   .gitignore
│   compile_commands.json
│   platformio.ini
│
├───include
│       README
│
├───lib
│   │   README
│   │
│   └───test
│           test.h
│
├───src
│       main.cpp
│
└───test
        README
```
with `lib\test\test.h` as 
```cpp
#ifndef _TEST_H_
#define _TEST_H_

#define MY_TEST 123

#endif
```
and `src\main.cpp` as 
```cpp
#include <Arduino.h>
#include "test.h"

void setup() {}
void loop(){ delay(MY_TEST); }
```
And I look into the generate compiledb, I find that the commands have the expected

```
-Iinclude -Isrc -Ilib\\test
```
switch in them.

jiriks74, 2022-10-06:
But why? I have an entire project that I've done in vsscode and it grows errors all over the place when I use an editor with clangd.

Yes the project is quite badly done, but I include a header in `lib)backend/eeprom.h` with just `#include "eeprom.h"` and it builds. When I open the project in editor that uses compiledb it throws errors all over the place even though PIO builds it with that no problem. 

I'll post you the tree and a part of the codewhen I get toy PC, if you want.

maxgerhardt, 2022-10-07:
>Yes the project is quite badly done, but I include a header in lib/backend/eeprom.h with just #include "eeprom.h" and it builds. When I open the project in editor that uses compiledb it throws errors all over the place even though PIO builds it with that no problem.

And you have 100% verified that there is no `-Ilib/backend` flag in the `compile_commands.json`? Can you upload the project and `compile_commands.json`?

jiriks74, 2022-10-07:
> And you have 100% verified that there is no `-Ilib/backend` flag in the `compile_commands.json`? Can you upload the project and `compile_commands.json`?

I have 100% verified it. I've searched for `-Ilib/backend`, `-Ilib` and even just the folder names `backend` and `lib`. Nothing. I am not sure, if I can send you the project, but I've created multiple new ones with some template header files with the same result, so I could definitely send you some of those when I get home.

LeCyberDucky, 2022-10-09:
I believe I'm running into a similar problem. 

I just installed clangd in VS code and created a new PlatformIO project. I then created `include\components.h` and defined a simple `struct` in this header. When I include this header in source files and use the struct, I get errors such as `use of undeclared identifier`. I see no mention of this header in the generated `compile_commands.json`. 

Is this the intended behavior? Am I perhaps missing a step when creating my header file, or am I placing it in the wrong spot?

jiriks74, 2022-10-09:
@LeCyberDucky 

If you haven't, try running `platformio run -t compiledb` to generate a new `compile_commands.json`. If this doesn't help, then add `build_flags = -Ilib -Iinclude` to your environment in `platformio.ini`. Then run `platformio run -t compiledb` again and it should be working for you

LeCyberDucky, 2022-10-10:
@jiriks74 Hmm, I tried that, but my `compile_commands.json` still ends up making no mention of `Project\include` or `Project\include\components.h`.

jiriks74, 2022-10-21:
Here is the project:
[project.tar.gz](https://github.com/platformio/platformio-core/files/9838263/project.tar.gz)

It should contain everything. Even a build when I tried if the project builds without issues.

When you open `main.cpp` the line `#include "test.h"` shows an import error but the project still builds successfully. The `test.h` file is in `lib/test/test.h`. I've tried it in `lib/test.h` as well and it still showed an import error.

ivankravets, 2023-04-21:
PlatformIO Core does not add `'lib` folder automatically to the `CPPPATH`. If you need it, please use `build_flags = -Ilib`. See https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html

closed 2023-04-21 by ivankravets