platformio/platformio-core #4438: Upstream udev rules

jurgelenas, 2022-10-14:
ExpressLRS is a heavy user of Platformio. We got a suggestion to upstream platformio udev rules to the systemd repository. Please see the quote below. I think Platformio organization is a more appropriate actor for this action.

> Hey as far as I understood the right thing to do with those udev rules would be to upstream them. So they wouldn't be needed here anymore.
> 
> The correct way to do that seems to be to add the devices to this [systemd/systemd@main/hwdb.d](https://github.com/systemd/systemd/tree/main/hwdb.d?rgh-link-date=2022-07-27T07%3A56%3A57Z)
> 
> I'm not sure if it fits in one of those files. Otherwise you would need to add a new one, here's an example PR how to do that. [systemd/systemd#22730](https://github.com/systemd/systemd/pull/22730)

Also, tagging @razzeee.

ivankravets, 2022-10-31:
Thanks for the issue. What should be done from our side?

AsciiWolf, 2026-01-19:
Any progress? :)